miegl/PiFmAdv #3: Less signal strength?

Niall7459, 2017-04-01:
It appears that the signal strength lower than PiFmRds
I used to be able to hear a few hundread meters away but now only like 50 meter with PiFmAdv

Miegl, 2017-04-01:
Can you try adjusting the deviation with `-dev` ?

Niall7459, 2017-04-02:
what do I put for -dev

Niall7459, 2017-04-02:
I set deviation to 25. What was it originally.

Niall7459, 2017-04-02:
I caanot test until we go in the car.

Miegl, 2017-04-03:
Its pretty weird that you got so bad signal compared to PiFmRds. I didn't change anything that could reduce signal that much. What RaspberryPi are you using?

Niall7459, 2017-04-06:
RPi 2

Its fixed, I changed frequency and -dev

reaches same amount if not further.

If i set -dev higher will this boost the range even more

Miegl, 2017-04-06:
Ok, glad its sorted out. Closing.

closed 2017-04-06 by Miegl

Miegl, 2017-04-16:
So, in addition to this; I went outside (lol) and tested the range of PiFmRds and PiFmAdv. Parameters were at their defaults.
And... the range its pretty much the same.
The deviation is set to 75 kHz as it is standard, and also to be able to fit in my country's limitations.

Niall7459, 2017-04-22:
? So if I CHANGE the Deviation

IT COULD GO FURTHER?

Niall7459, 2017-04-23:
????
Please reply @Miegl 

If increase deviation will go further.
I expect not.

Miegl, 2017-04-23:
Well basically yes. But you can expect the audio quality to be worse.

Niall7459, 2017-04-24:
At default deviation full bars
At 100 deviation is full bars
at 280 Deviation is 4/10 bars.

What sort of increase can i expect by increasing deviation?
cntermetres meters KILOmeters?

Miegl, 2017-04-24:
Well deviation != distance.
Deviation is basically bandwidth. Thats why low amount of deviation will result in less distance. And hude amount of deviation isnt good too.

I personally recommend setting the deviation 50 - 75 kHz. I would nener set the deviation to more than 125 kHz.

And of course this will not result in kilometer increase. I don't know what antenna are you using, but Raspberry Pi itself generates very very low signal.

Niall7459, 2017-04-24:
Meters increase

Niall7459, 2017-04-24:
My -dev is currently 150 full bars on the radio a few metres away, quality is noticably worse.

Niall7459, 2017-04-24:
Does deviation at like 300, this wouldnt effect RDS would it??? Make the RDS less strength?

At my freinds house on his radio IT can pick up the RDS but sound is static.

Miegl, 2017-04-24:
Yeah. All the stuff that gets modulated (mono sound + stereo + pilot tone + rds) is also affected by deviation. So as I said; its better to set deviation at cca. 70kHz and use better antenna, which will affect the distance the most.

Niall7459, 2017-04-29:
@Miegl . I have my antenna as peice of wire in upstairs of my house, how to build better antenna.

Miegl, 2017-04-29:
@Niall7459 google it maybe?