miegl/PiFmAdv #5: external exciter compatibility + rds and lowpass filter ciruit

Niall7459, 2017-04-22:
Can I hook up to an external Transmitter somehow

I want to boost range.

And if i was too would RDS work?

****** Considering the transmitter has Low-Pass, will this filter out RDS???? ***

Miegl, 2017-04-22:
Haha, this won't work.
If you want to boost range of PiFmAdv, use better antenna and amplify the signal with some fm amplifier.
If you want to use "real" fm transmitter and generate your own RDS data via soundcard, then you must make sure that the fm transmitter doesn't have low pass filter, which will filter you RDS (and other sub-carriers).

Niall7459, 2017-04-23:
I will generate RDS using a RDS encoder.

This setup OK?? ::
             
RASPBERRY             ------>    FM TRANSMITTER -------> RDS ENCODER ------->     Antenna
PI (Plays Music)

Miegl, 2017-04-23:
Well Raspberry Pi has crappy audio output (its just GPIO), so I highly recommend using some external audio card.
And yes, this would work. (I will rather get fm transmitter with rds encoder built in tho)

closed 2017-05-07 by Miegl

Niall7459, 2017-11-08:
@Miegl Do you know cheap FM transmitter with RDS? I want about 0.5 Watt transmitter

Also I looked at instructions for the rds encoder and you cant 
FM TRANSMITTER -------> RDS ENCODER -------> Antenna

it is like

                         RDS ENCODE
                                     |
   AUDIO ---->   TRANSMITTER -----------> Antenna

Miegl, 2017-11-09:
Just search for it on the internet. Its offtopic to this project.
And yea, mistake on my side, sorry for that. You need to generate RDS signal before transmitting it, which means that you either have to add RDS to the signal before entering the transmitter using RDS encoder or you can simply play your sound and also provide RDS signal with software such as the RDS encoder inside PiFmAdv.

Niall7459, 2017-11-12:
Hello i have one last question. I have found two transmitters.

A signstek 0.5 W transmitter NO RDS (It has lowpass so I cannot generate RDS by software) 50$
OR
A 0.2 W Vastelec Low Power FM Transmitter With RDS Encode. 120$

Im not really sure what to do... Because I want a good range, but I also want RDS and I dont want to go over 500mW

Niall7459, 2017-11-12:
Also how much further will 200mW go than Raspberry Pi?

Miegl, 2017-11-12:
Well according to maximal drive strength 7 = 16mA, power output of the Rpi would be 3.3 * 16 = 52.8mW.
But since the Rpi isn't transmitting only on one specific frequency (mesh etc.) the power output on specific freq would be way lower.

Niall7459, 2017-11-12:
Okay Im going to buy the 0.5 W transmitter but it has no RDS.

So it would be great if you could add AF to PiFmAdv so when they loose the Pi station will switch to my main transmitter.

Thanks!

Miegl, 2017-11-12:
Be aware that for this to work your main transmitter has to provide RDS signal with the same PI.

Niall7459, 2017-11-12:
oh..

Niall7459, 2017-11-12:
My other transmitter has stupid LOW PASS FILTER

Miegl, 2017-11-12:
Well you can try to build your own. You can start off by building a fm transmitter with only one transistor. You can then provide your audio + rds signal from let's say pc soundcard. The soundcard has to support 192khz for rds to work tho. 96khz is enough for stereo.

The only problem with one transistor fm transmitters is that the stability of the frequency is a bit awful.
But again, it's very easy to build one off the shelf.

Niall7459, 2017-11-12:
I think I will just buy one it looks nice in my roomlol