miegl/PiFmAdv #9: Crashes on pi3

Niall7459, 2017-10-31:
When I start the program it crashes my pi..

I cannot VNC or anything but it doesnt shutdown it just freezes.

Niall7459, 2017-10-31:
Program loads and everything, I can hear the radio but cannot access pi.
This worked okay on mypi 2

Niall7459, 2017-10-31:
I am using VNC but i also cannot FTP or SSH onto the pi in this state.

Will a apt-get dist-upgrade fix this....?

Niall7459, 2017-11-02:
When i move a window while the TX is running it freeze.s

Miegl, 2017-11-12:
Hm... weird. I don't have the Rpi 3 so I can't really test it, but maybe it's an issue with DMA channel?
You can try to apply this commit to PiFmAdv: https://github.com/SaucySoliton/PiFmRds/commit/fcfccdd0fe06c556f9b4e5776b516c1b9f5f2093

If the problem persists try running PiFmAdv on bare Arch Linux Arm (without xorg etc)

Niall7459, 2017-11-12:
I run it headless now anyway

Niall7459, 2017-11-12:
okay @meigl Is there a command I can use to apply this commit?

Niall7459, 2017-11-12:
Never mind its only small

Niall7459, 2017-11-12:
Do i have to MAKE?

Niall7459, 2017-11-12:
OH MY YOU LEGEND IT WORKS GREAT THANK YOU THANK YOU THANK YOU THANK YOU NOW I CAN USE THE GUI!!!!!

Question : this doesnt affect performance using DMA 5?

Miegl, 2017-11-12:
No, this doesn't affect performance.
I'll soon commit new update but I need to do some testing first.

Niall7459, 2017-11-12:
Cool thanks it works great.
I'm now doing some cool testing with the fifo pipe.

Question: what is best antenna to use? homemade like a 1/2 wave dipole?

Niall7459, 2017-11-12:
OH no problem My audio play slightly fast and cuts out after a few seconds!

Niall7459, 2017-11-12:
Okay I have now set define DMA_NUMBER 6 And it seems to work okay

Miegl, 2017-11-12:
well DMA channel 6 is reserved by GPU so idk how that will work out. You can try any of these channels which appears free: 4, 5, 8, 9, 10, 11, 12, 13, 14

Niall7459, 2017-11-12:
oh lol 6 is working fine at the moment

closed 2017-12-18 by Miegl