miegl/PiFmAdv #10: Recent PiFmAdv changes

Miegl, 2017-11-12:
Hello,
I just pushed a pretty big change to this project.

PiFmAdv is now modulating the PLLC instead of clock divider. This has a big impact on quality. The stereo reception is now much better. Also stations on close frequencies to PiFmAdv aren't jammed and can be listened to even in close range to PiFmAdv antenna.

This is all great but this change has also some negative impact.

Firstly you need to add `gpu_freq=250` to `/boot/config.txt` to prevent frequency stability issues.

It would be much appreciated if you guys would test it out and post feedback on this issue!

preterive, 2017-12-03:
I update now and then i test the new one

preterive, 2017-12-03:
The sound quality is better, but i send on 88,70 and i can't listen 89,00 if PiFmAdv is on.
I use raspberry pi 1 B

Miegl, 2017-12-15:
@preterive would you be kind enough to share your enhancements with other or maybe make a pull request? thanks!

Niall7459, 2017-12-16:
Hmm okay this looks good im going to test it.

Niall7459, 2017-12-16:
Could you help me? 

Im using arecord to record from Loopback devices to STDIN . but I am getting OVERRUN!! errors and slow audio.

This happened on the old version as well.

There is also high ptiched noise but Im not sure if this is my Arecord problem.

Niall7459, 2017-12-16:
My Clock speed is 900000

Miegl, 2017-12-16:
@Niall7459 the high pitched noise can be the stereo pilot tone or subcarriers depending how much it is slowed down.

I was getting slow audio after relaunching PiFmAdv multiple times too, but reboot fixed it.

Niall7459, 2017-12-16:
@Miegl Same, after a reboot that would fix the problem, but no more for me..

Anyway just discovered that my pi 2 has died halfway through installing ffmpeg.
Ive had it running 24/7 for a good year... or so
Il try fix it or hook it up to monitor maybe it needs a new SD card.

Niall7459, 2017-12-16:
Oh and @Miegl Does this new system make the reception range further? that would be great!

Miegl, 2017-12-16:
@Niall7459 lol the same thing happened to me recently... RIP sd card. And no, the reception is about the same, it really isnt possible to make it better without hw mods.

Niall7459, 2017-12-16:
Hw mods? What are they? And any tips for improving range??

Niall7459, 2017-12-16:
Hope fully my sd card just needs reformatting

Miegl, 2017-12-16:
@Niall7459 you can use some hat with amplifier. I've seen these for the 4m and 2m band so it should be easy to make one for FM.

Niall7459, 2017-12-16:
Where I can buy this??

Niall7459, 2017-12-16:
oh well, downloading noobs now. gonna reinstall.

On the upside: At least this fixes my problem with slow audio (hopefully)

closed 2018-04-23 by Miegl