miegl/PiFmAdv #11: Pifm zero

preterive, 2017-12-16:
Hi,
is it working on a pifm zero?

Miegl, 2017-12-18:
Haven't tested, but it should. If something goes wrong share your feedback please. Thanks.

closed 2017-12-18 by Miegl

Stealthbird97, 2018-12-04:
I've been trying to get this to work on Pi Zero W. The program seems to work, unfortunately, the Pi doesn't like something. It will transmit, however, the Pi basically becomes unresponsive afterwards. The WiFi dies, the camera doesn't respond etc.

anducampu, 2018-12-16:
I have the same problem exactly on pi 0w. However on pi 3b works fine.

arhico, 2018-12-16:
Have the same problem on my zero W. Commented out gpio 34 release to get it to work ok.

mrphax0r, 2019-08-21:
It works, tho with minor stuttering.

mrphax0r, 2019-08-21:
I suspect that it overheats since I applied a heat sink