miegl/PiFmAdv #12: Console have no responce after running it

wlmqljj, 2017-12-16:
Model: Pi 2b
System: Rasbian stretch and the newest version in November 2017
stuck and cannot exit after running
--------------------------------------------------------------------------------
root@raspberrypi:~/PiFmAdv/src# ./pi_fm_adv  -audio sound.wav
Carrier: 107.90 Mhz, VCO: 971.1 MHz, Multiplier: 50.578125, Divider: 9
Using mbox device /dev/vcio.
Allocating physical memory: size = 4354048, mem_ref = 5, bus_addr = fe6c8000, virt_addr = 0x767d3000
PPM correction is 0.0000, divider is 2129.6052 (2129 + 2479*2^-12).
RDS Options:
RDS: 1, PI: 1234, PS: <Varying>, PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 107.9 MHz.
----------------------------------------------------------------------------
And no audio output hearing in the radio.Then I used my rtlsdr to check the spectrum , pi only  transmitted just a stright line on FFT output. But when i use the old version of rds.c committed on13 Nov. And after compiling ,it works nice .I dont know what happened ,maybe the DMA's problem?
Thank for your such intersting software.

Niall7459, 2017-12-17:
I can confirm this happened to me. My VNC Server froze up, Music was still playing through the radio though, and then It started skipping... (I use avconv) And I had to pull the plug on RPI.

Could just be a coincidence, my pi locked up the other day..

Miegl, 2017-12-18:
Well I never had a problem running it headless. I'll try to spot and fix this issue once I get a new sd card for my rpi.

Niall7459, 2017-12-18:
Il try without booting without graphical interface next time.
And yes it has just frozed up again.

Miegl, 2017-12-18:
Freezing up has to do probably with either PLLC being altered or DMA.

Miegl, 2017-12-18:
I have developed this under Arch Linux Arm, so theres a great possibilty something in Raspbian is interferencing.

Niall7459, 2017-12-19:
Ok so on raspberry pi 2 what DMA Channels are free?

Miegl, 2017-12-19:
4, 5, 8, 9, 10, 11, 12, 13, 14
I can't test the mright now. It's possible that something is using those lower ones tho, but the higher ones should work without a problem.

Niall7459, 2017-12-22:
I run the PiFMADV via SSH.
Works fine.

Then I log into VNC. If i open an application the PIFmAdv Breaks and I have to restart my pi to get it back..

Miegl, 2017-12-22:
@Niall7459 yeah that's sort of expected. Don't know if I'm able to fix this though. But I can't really see a scenario where I need to use PiFmAdv and DE at the same time, so it really isn't a big problem.

Niall7459, 2017-12-22:
@Meigl Ok

Is there ANY way I can boost the range just slightly??? Like without extra equipment.

Although I have some spare Wire, I could build  a dipole?
How much better is a dipole?

Niall7459, 2017-12-22:
@Miegl sorry spelt it wrong see message above

Miegl, 2017-12-22:
@Niall7459 well you can get better range with using low freq (and an antenna fir that freq)

Niall7459, 2017-12-22:
@Miegl how much better would it be?

Niall7459, 2017-12-22:
Metres, 10s of meterS?

Miegl, 2017-12-22:
@Niall7459 Don't know. This is pretty off topic tho and I'm probably not the best person to answer that.

Niall7459, 2017-12-22:
ok

Niall7459, 2017-12-22:
@Miegl when i have no audio i can hear very quiet, but high ptiched tone. (as you said with slow audio)

Does this mean calibration might be out?

Niall7459, 2017-12-22:
Also after a 30 mins or so the Radio just cuts out.
Pi Fm does not stop though...

Niall7459, 2017-12-23:
@Miegl by "headless" does that mean just without monitor and via ssh or something.. Or do you mean that the pi is never booted into GUI ?

Miegl, 2017-12-23:
@Niall7459 by headless I mean no GUI running. Meaning the Xorg server (or wayland) is not running.

Niall7459, 2017-12-28:
I reformatted and got the new pixel interface. I do not like it and want to remove it

Miegl, 2017-12-28:
Ok so I found out that PiFmAdv behaves differently on different distros. I think it's due to the kernel version but I'm not sure. I will do some testing on different distros and kernel versions.