miegl/PiFmAdv #13: I have a small question about altitude of transmitter.

Niall7459, 2017-12-23:
have a question but had to make issue , sorry.

lets say my house.
_.====._  
|             | 
|________|
|             | <------ Transmitter downstairs (Worse reception)
But then...
_.====._  
|             | <------ Transmitter Upstairs (I get much better reception)
|________|
|             |

But this is slightly inconvenient and I would like to possibly have the Raspberry Pi Downstairs.
Can i have a wire traveling up the side of my house. and expect the range to be the same as if the Pi were upstairs.

_.====._  
|          ( ( |(+ ) )  )     )  <--- End of wire and then radio waves.
|________|  |<---- Wire from RPI 
|         Pi |-|     <-----Raspberry pi downstairs

Because surely there would be waves generated throughout the height of the wire. up the house.
So intern there would be less waves generated at the top..

_.====._  
|             | + ) ) ) ) ) ) ) ) ) ) ) <-- radio waves generated at end of wire.
|________|  |<--- wire
|             |-|<--- Rpi

Thats what i want cos more radio waves generated at higher altitude.

_.====._  
|             | + )}}} less waves at top 
|________|  | }}}}
|             |-| }}}} - Waves generated from all parts of wire!

If you know anything about this that would be great.

Miegl, 2017-12-23:
Sure, you can just use coaxial cable that then connects to an antenna or something.

closed 2017-12-23 by Miegl

Niall7459, 2017-12-23:
Oh those drawings didnt work out did they

Niall7459, 2017-12-23:
Will 87.7 Mhz go further than 107.9 Mhz?

Miegl, 2017-12-23:
Again, you have to use coaxial cable instead of normal wire to get the signal upstairs. And yes, technically 87.7 Mhz will go further. (But you have to use antenna for that specific freq)