miegl/PiFmAdv #15: Monitor had an issue and then the Pi just hung.

MrSchumck, 2018-01-01:
Upon starting this program, the monitor i was using started showing white and black dots all over the place, then my Raspberry Pi 3 stopped responding and i had to reboot. tried again and had the same effect.

Miegl, 2018-01-02:
Please try out my latest commit https://github.com/Miegl/PiFmAdv/commit/05f63449928c77d0609611787f4faa2df75a725e and let me know if the behaviour differs. Thanks!

MrSchumck, 2018-01-02:
okay so that got rid of the monitor behaviour but now the FM station stops and skips and stutters then it shuts off completely.

Miegl, 2018-01-02:
What distro & desktop enviroment are you using?

Niall7459, 2018-01-10:
Hi im now getting overrun!! again :( I cant seem to target what causes this..

Niall7459, 2018-01-10:
Il try in CLI