miegl/PiFmAdv #16: rds 2.0

preterive, 2018-01-07:
Hello,
when comes rds 2.0 ? 
These are the features:

    Seamless support for frequencies from 64 MHz to 108 MHz (AF, EON)
    New character coding: UTF-8 (old EBU Charset remains for compatibility mode for the old 0A/2A Groups).
    New ODA handling, "B" groups are assigned as signalling group to the “A” groups.
    Long PS-Name, up to 32 byte with UTF-8 character set. (Indian, Chinese, Arabic, and more)
    RadioText (eRT) 128 byte long with UTF-8
    Increased capacity from 11.4 up to 57 "A"-groups per second. (2,109 bit/s. net capacity with the single modulation-type multiple subcarriers (SMMS) technology)
    Graphical RadioText – supports HTML/CSS templates (for smartphones, car radios, computers/tablets)
    Supports return channel over gRT if the receiver has IP or SMS capability.
    Broadcaster's graphical logo – a maximum 4 kilobyte picture (JPEG, PNG, or GIF)
    Hybrid Radio feature (partly based on Radio France development)

I really like this.

Niall7459, 2018-01-10:
cool

Miegl, 2018-01-10:
Hey, it's possible to implement xRDS.
The main problem is that I don't have a single receiver that can actually use xRDS.
And from my point of view xRDS isn't even used. All the big FM stations in my country are using just basic RDS services and nothing more (Some of them use RT+ tho).
And to be fair, I don't think it will get better. FM is slowly dying and even now it is pretty much replaced with internet.

So unless someone makes a PR, I don't guarantee I will ever implement this.

Niall7459, 2018-01-10:
Ok. Also @Miegl  I really need some help do you have skype or something I can speak to you?

Miegl, 2018-01-10:
@Niall7459 you can contact me via email (my github page). Oh, and you can use PGP if you wish :)