miegl/PiFmAdv #18: Best antenna to increase range?

Brandon4466, 2018-01-15:
Are there any suggestions for an antenna to use with this? Other's experiences?

I have found something like this: https://www.amazon.com/dp/B01HTDKAWS/ref=cm_sw_r_cp_apa_JsbxAbGW04SRV

Would something like this work? Does anyone know how you would connect something like this best to the GPIO?

Basically just the best way to boost the radius. Any suggestions would be great! What do you guys use for an antenna?

Thanks!

sm3ulc, 2018-01-15:
> TDKAWS/ref=cm_sw_r_cp_apa_JsbxAbGW04SRV

If you plan to anything else than a piece of string for short term use you
should add a LPF such as the one below.

https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F282734932208

With that said the stick will work ok. I tried a diamond x200 and got some
500m range with RDS.

// David

Stinkers, 2018-01-15:
Be aware that driving an inductive load (such as a filter) directly could damage the Pi's output hardware. I'm currently working on a simple circuit that should be provide a nicer load for the Pi and filtering. The plan is to turn it into a hat.

Miegl, 2018-01-17:
You can get a decent range with just a simple dipole. But even then the range won't be very good. That's because the power output of the RPi is low. With a good radio reciever you will be able to listen pretty far away, but other recievers will have problem picking up the signal even at a small distance.
It would be possible to design a hat with both filter (band pass) and small amplifier.

Niall7459, 2018-01-21:
I use a wire about 60cm long, It works quite well can pick up good range in a car. but no so much with other recievers, also try placing the Pi upstairs that increased my range by 2x and also try finding a frequency that is clear from interruption because I found this reduced the distance.

closed 2018-01-26 by Miegl