miegl/PiFmAdv #19: How can I make cars switch to my station

Niall7459, 2018-01-20:
I want cars to be able to switch to my station maybe using some combination of RDS TA and EON on other stations....
Or maybe rds EAS pty 31 the Emergency Alert System but I think it only works in America.

So basically I want the cars to switch to my station as soon as there in range.
Just for testing thanks

Miegl, 2018-01-20:
You can't make a car switch to your station (thankfully).

The closest thing possible is to transmit on a freq that is in AF of some other station and as soon as that station looses signal the cars tuned to that particular station will (or should) switch to your station. For that to work though your stations PI must be matching the PI of the station.

But for that the other station must loose its signal (or at least the RDS reception has to get worse), which will probably involve jamming. And no, I wouldn't do that, even for "testing" purposes.

closed 2018-01-20 by Miegl

Niall7459, 2018-01-21:
im sure it would be fine

what the worst that can happen lol ?

ofcom wuld never no

Niall7459, 2018-01-21:
I want to build a relay of transmitters like so 1 pi transmits then PI2 recieves and re-transmits

HOME ) ) ) ) ) ) ) ) ) PI ) ) ) ) ) ) ) ) PI ) ) ) ) ) ) ) PI ) ) ) ) )))

Niall7459, 2018-01-21:
But I need something to recieve FM for the other pis, and power and that would be cool
Except music might be out of sync slightly.

Miegl, 2018-01-21:
I would you use wifi ptp for that

sm3ulc, 2018-01-21:
The normal way is to use RDS and AF-lists (alternative frequencies). Check
the RDS specification and add some rows to transmits the extra RDS-groups.

// David

Miegl, 2018-01-21:
@sm3ulc Yeah I will add AF. This will not make the cars switch to a PiFmAdv station though, as @Niall7459 wants.

Stinkers, 2018-01-21:
On Jan 21 2018, Josef Miegl wrote:

> @sm3ulc Yeah I will add AF. This will not make the cars switch to a 
> PiFmAdv station though, as @Niall7459 wants.

https://en.wikipedia.org/wiki/Traffic_announcement_(radio_data_systems)

### Changing PS, RT and TA at run-time

You can control PS, RT and TA (Traffic Announcement flag) at run-time using 
a named pipe (FIFO). For this run Pi-FM-RDS with the `-ct l` argument.

Example:

```
mkfifo rds_ctl
sudo ./pi_fm_rds -ctl rds_ctl
```

Then you can send "commands" to change PS, RT and TA:

```
cat >rds_ctl
PS MyText
RT A text to be sent as radiotext
TA ON
PS OtherTxt
TA OFF
...
```

Miegl, 2018-01-21:
TA will not make a car switch to your station from another station. It will only switch to your station if you are tuned to your station but using lets say CD player.

sm3ulc, 2018-01-21:
Some documentation about the RDS system for reference:

https://www.iz3mez.it/wp-content/library/ebook/RDS%2520-%2520The%2520Radio%2520Data%2520System.pdf&ved=2ahUKEwiguNuA3OnYAhXEkiwKHajgBssQFjAJegQICBAB&usg=AOvVaw0v97QQ3UraQUq5pV3Dnz7W

Kind regards,

// David

sm3ulc, 2018-01-21:
Yes and switching to another fq on TA is with EON. Not to confuse. :)

// David

sm3ulc, 2018-01-21:
> 2520-%2520The%2520Radio%2520Data%2520System.pdf&ved=
> 2ahUKEwiguNuA3OnYAhXEkiwKHajgBssQFjAJegQICBAB&usg=
> AOvVaw0v97QQ3UraQUq5pV3Dnz7W

Pretty broken url, trying again:

https://www.iz3mez.it/wp-content/library/ebook/RDS%20-%20The%20Radio%20Data%20System.pdf

// David

sm3ulc, 2018-01-21:
Also, maybe worth to add in documentaion. This easy software by Oona is
very good for decoding/testing/analyzing RDS:

https://github.com/windytan/redsea

// David

Miegl, 2018-01-21:
@sm3ulc thanks for your recommendation. I have to (finally) buy some cheap rtl-sdr ;)
There's also another useful tool called RDS Spy (http://rdsspy.com/).
It's for Windows, but it runs jsut great under wine.

Miegl, 2018-01-21:
Ok, so finally I have done implementing AF ;) https://github.com/Miegl/PiFmAdv/commit/25f7dc5b556c4009d7b64f7e8f8c05410384f5dd

[image]