miegl/PiFmAdv #22: Audio frezzes after a few seconds

Pielke, 2018-01-29:


Miegl, 2018-01-29:
Well, obviously by closing your terminal the application exits too. I need better report in order to fix the issue you are having.

Pielke, 2018-01-29:
I start the program like any other version of the PiFm based programs... like sudo  ./pi_fm_adv --audio sound.wav -- freq 107.9 and so on. After starting the program it works only for a few seconds, then the audio stops playing but the text in the terminal stays. Then it lasts a few seconds after pressing ctrl +c to kill the process. Same thing, when entering (not closing the main terminal) an other terminal e.g. for using the ta and rt functions.

Miegl, 2018-01-29:
That looks like a issue with gpu and pllc... Try exiting out of gui and try again please.

Pielke, 2018-01-29:
Do you mean to start the program in cmd mode of the pi?

Pielke, 2018-01-29:
Same thing... Audio just crashes or stucks in an infinite loop of just a few milliseconds

Miegl, 2018-01-29:
Please post the output of `uname -a`

Pielke, 2018-01-29:
Can I do it tomorrow... Or in 2-3 hours...

Pielke, 2018-01-30:
output of uname -a:
Linux raspberrypi 4.9.59-v7+ #1047 SMP Sun Oct 12:19:23 GMT 2017 armv7l GNU/Linux

Miegl, 2018-01-31:
Hm, interesting. It's probably kernel (idk for sure) related. However, PiFmAdv (as of now) works fine under the latest Arch arm and Ubuntu server. So, if possible, try updating your distro/kernel.

Part of the code has been changed to work on latest distros though. I've commented out some parts of the code. So in order to get PiFmAdv running under your system, you can try making these changes:
In `pi_fm_adv.c`:
Remove `//` on lines `282, 283, 284, 285, 295, 296, 310`
That should be enough to get it working for you. I don't know why some distros (or kernels) require these additional lines to get PiFmAdv working but the current version works fine under latest Raspberry Pi kernel.

Pielke, 2018-02-01:
Thank you, I will try it as soon as possible...

hectorperalta96, 2018-02-02:
Hi, I have the same problem. Audio works fine for few seconds and then it stops playing. The text stays on the terminal and the carrier stays too. I don't have any GUI installed on the Pi.
I tried to remove `//` on lines you said but still having the same problem.
I'm using raspbian and the output of my `uname -a` is:
Linux raspberrypi 4.9.79-v7+ #1086 SMP Wed Jan 31 18:03:34 GMT 2018 armv7l GNU/Linux

Miegl, 2018-02-05:
Well, this looks like a complicated issue. I've noticed the same problem, but I just adjusted the code everytime it broke. (I'm using Arch Linux ARM on rpi2)
I think this has to do with kernel, but I'm not entirely sure. Any suggestions are welcome.

Miegl, 2018-02-15:
So, somehow this affected me too and the desired transmission last only for a few seconds. Then the frequency completely shifts away and creates horrible mess. It's really weird, cause it worked for me for very long time and now it suddenly doesnt...

Pielke, 2018-02-17:
I would guess, that this problem is caused by audio processing. Different wav's have different playing durations before the system collapses. Can it bei the caching of the Audio/ wave-data?

Pielke, 2018-02-20:
Thank you for your response! Where can I get the older Firmware Versions from?

Miegl, 2018-02-22:
But I didn't quite fix the issue. Even with the older firmware the sound is "cracking". And the gpu/pllc will still switch to a different freq after some time. The worst part about this is that it happened to my Raspberry Pi 2 randomly, not package upgrades etc. It was working fine, and now it's just not... I don't know, the PLLC method is far better quality-wise but it just doesn't work as good.

Pielke, 2018-02-26:
We just need it to work for a short time Period... So this would be OK. But I don't know how to download the older Firmware Version.

TheSneakySniper, 2018-02-28:
@Miegl 

> Hm, interesting. It's probably kernel (idk for sure) related. However, PiFmAdv (as of now) works fine under the latest Arch arm and Ubuntu server. So, if possible, try updating your distro/kernel.

What version of Ubuntu server is confirmed working with the raspberry pi?

hectorperalta96, 2018-02-28:
As far as I'm using the last rpi firmware, older versions of PiFmAdv still working for me, for example, this one:
https://github.com/Miegl/PiFmAdv/tree/f2968ca362c0a3dbfea31a1484a4063bc9176717

No audio freezing.

TheSneakySniper, 2018-02-28:
Using ubuntu server ARM os, it allows me to stream audio using this application, however, it is much less powerful with ubuntu server than with raspbian 

Not sure why this is.

Miegl, 2018-03-03:
@hectorperalta96 thats because it was using PLLD, which just isnt as problematic. I might add PLLD method back alongside PLLC.

Miegl, 2018-03-03:
@TheSneakySniper Ubuntu server 16.04.4 is working fine

TippyLion28, 2018-03-31:
Dunno if this is related but the SD activity light on my Pi goes out immediately once the audio stops

TippyLion28, 2018-03-31:
I might just be going insane but it seems mounting all filesystems as read-only using Alt-SysRq+U fixes everything after running the program 1-2 more times. What the hell xD

Niall7459, 2018-04-24:
you could try sudo rmmod w1-gpio
That fixed the transmission stopping for me.

Is it just silent or goes static?