miegl/PiFmAdv #24: Program malfunctioning on most recent version of Raspbian GNU/Linux 9

TheSneakySniper, 2018-03-01:
cat /etc/os-release
`PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"`
uname -a
`Linux streampi 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l GNU/Linux`

When I try to run the program, it first crashes. Then running it again causes it to work for 15 seconds, then stop working.

When the GPU frequency is changed to 250 (or any number) it broadcasts nothing indefinitely and hangs the pi. I can not control-c the program. 

Here is the weird thing.

I installed ubuntu server onto my raspberry pi's SD card, and it seems to have fixed all of the issues i was having, and allowed me to broadcast audio perfectly. However, the range was severely limited to a couple feet with and without an antenna. Before I was getting ~100 feet with a jumper wire attached, and now I only get 3 feet (even with -w set to 7!)

What OS can I install to make this work with a long range and allow me to broadcast for more than 15 seconds?

Miegl, 2018-03-02:
Oh well... Yeah it doesn't really work as expected now...
However I was running it on Arch Linux Arm without an issue. (Then it well stopepd working out of nowhere)

TheSneakySniper, 2018-03-02:
@Miegl Do you know what happened that caused things to break? 

I am actually very interested.

Miegl, 2018-03-02:
@TheSneakySniper the thing that causes things to break is the gpu (or PLLC). Basically other pifm like programs use PLLD, which is working fine but the signal quality (and purity) is bad and you can't get even good stereo reception (there's a lot of noise). PLLC modulation puts the quality and purity on a whole new level with decent sound and even stereo reception is okay. Well the downside of it is that while it worked for me for some time it just breaks a lot. I'm pretty sure it has to do with the raspi firmware. Maybe a  open source firmware could solve this. I don't know.

TheSneakySniper, 2018-03-03:
@Miegl Would UART being disabled by default in the most recent version of the kernel have anything to do with this issue?

> For some bizarre reason the default for Pi3 using the latest 4.4.9 kernel is to DISABLE UART. To enable it you need to change  enable_uart=1 in /boot/config.txt.

Miegl, 2018-03-03:
@TheSneakySniper I don't own rpi 3 or zero w so I can't this. However, I think that it has no impact.

Miegl, 2018-03-03:
So I managed to get pifmadv to work. It's not a perfect solution tho, as I had to install Ubuntu server 16.04.4 (http://cdimage.ubuntu.com/releases/16.04/release/ubuntu-16.04.4-preinstalled-server-armhf+raspi2.img.xz). It's probably the old kernel or/and firmware thats doing the trick. On latest arch linux arm the gpu (pllc) is just behaving differently.

So yeah, just install Ubuntu server 16.04.4, git clone latest pifmadv and set `gpu_freq=250` in `config.txt`.

TheSneakySniper, 2018-03-03:
@Miegl Last time I tried pifmadv on ubuntu preinstalled server, the range was very, very limited from what was previously capable. Now it is only capable of 1-5 feet with or without an antenna, compared to 50-100 meters.

Miegl, 2018-03-03:
@TheSneakySniper thats honestly very weird.

TheSneakySniper, 2018-03-03:
@Miegl So you are using a rasberry pi 2, yes?

Miegl, 2018-03-03:
@TheSneakySniper correct.

TheSneakySniper, 2018-03-03:
@Miegl Let me try reinstalling it.

Niall7459, 2018-03-11:
I use raspberry pi2 as well, Old PiFmAdv (before PLLC) works altright but I cannot run with the GUI, it must be completely disabled or else i get skipping, slow audio and crashes.

I cant get the the PiFmAdv working at all on RPI2
Although i cant remeber if i set grpu_freq

JsBergbau, 2018-04-10:
@Miegl 
Can you tell us which driver exactly is making the trouble? You can download old images there http://downloads.raspberrypi.org/raspbian/images/ or http://downloads.raspberrypi.org/raspbian/images/raspbian-2017-12-01/ for the old version Stretch version from November 2017
Then you extract the drivers from the image and put them into the current raspi, then reboot and it should work again.

Have you already posted the problem in the official Raspi Forum? Perhaps it is just a bug, like they had with Wifi before see here https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=206083#p1279262
If we knew the package name, there is even no need to extract the drivers from the old image.

Miegl, 2018-04-10:
@JsBergbau last time I tried (~10 days ago) latest Arch Linux Arm is working fine. Don't have time to test now sorry.

JsBergbau, 2018-04-10:
I've tested today on Zero W and Stretch from 13.03.2018 without doing sudo apt upgrade. I've only tried for a few seconds, but it was able to mask the original radio station with a paperclip plugged through the GPIO hole, so there was no real direct or only slightly electrical connection. Radio was 2 m away. Putting my finger on the GPIO PIN without the paperclip led to strong noise (was trasmitting silence) on the radio. So transmitting power wasn't affected. I've tried with the "original" PiFMRDS, so I don't know if there is any difference to your version.  BTW do you know what transmit Power the PI reaches?

Miegl, 2018-04-10:
@JsBergbau @dabfree has measured the power output to be ~10mW.

Niall7459, 2018-04-10:
ok let’s try get that at least 100 W lol

JsBergbau, 2018-04-11:
> I will probably update readme with the actual differences at some point in the future.
Really curious to know the differences.

Since youre doing all the modulation stuff for RDS, FM and so on, would it also be theoretical possible to  modulate DAB+ via software and then send it out like FM?

Niall7459, 2018-04-11:
That sounds like a pretty cool idea and I would definitely use it. Although I believe it may travel less distance and I’m no expert but it may be easier to program since your just sending 2 different patterns representing 1 and 0. However i have a feeling that the raspberry pi modulation cannot modulate up to the DAB+ spectrum (I may be wrong) I remember checking it for something and it couldn’t modulate 868MHz 

And of course I think RDS would be the hardest part hopefully he can find a technical specification somewhere to implement it.

Overall a great idea but I don’t think it’s possible...

Miegl, 2018-04-11:
@Niall7459 I'm not sure if the GPIO pins are fast enough for DAB.

dabfree, 2018-04-12:
That isn't possible unfortunately. DAB is an OFDM system using 1536 DoQPSK
modulated carriers, each spaced 1 kHz apart. FM is just one carrier.

There is a tool on my github which can produce a DAB signal from a VGA
graphics card with the opendigitalradio.org toolset. It isn't perfect, but
it works.

R.

JsBergbau, 2018-04-13:
``That isn't possible unfortunately. DAB is an OFDM system using 1536 DoQPSK modulated carriers, each spaced 1 kHz apart. FM is just one carrier.

So if there are 1536 Carriers each 1 KHz appart it makes a total Bandwith of 1,536 MHz. So doing signal processing you can combine them to 1 signal (DSL modems work that way I think). Question is if PI is able to send out a signal with 1.536 MHz Bandwith. Or am I wrong? You're the signal expert, can you explain it more?

PS: Cool stuff with modulation DAB from VGA

TheSneakySniper, 2018-04-15:
Should I attempt running this on my raspberry pi 3 B+ with this os:
https://www.raspberrypi.org/downloads/raspberry-pi-desktop/

Or will it not work guaranteed?

morrolinux, 2018-04-24:
@Miegl I've got a Pi3 to try, I've installed ubuntu 16.04.4 (preinstalled image for pi3), and created a file  ´/boot/config.txt´ with ´gpu_freq=250´ inside. Problem is, with or without it, the system just freeze. I think that could be because the pi3 has a default core_clock at 400 and all the others at 300? Could you give me some advice here on how should I set the clock? where's exactly needed to be 250mhz?

Miegl, 2018-04-24:
@morrolinux more details? I have tested that PiFmAdv works fine now with latest Arch Linux Arm (thats what I use)

morrolinux, 2018-04-24:
I don't know, what info would you need? does gpu_freq needs to be set to 250 on the Pi 3 as well? I cannot get a Pi2 and just tried to install Ubuntu 16.04 as instructed above, but I hear no signal and the screen goes out (sometimes blank, sometimes white-dotted, depending on which frequencies i put in /boot/config.txt

Miegl, 2018-04-24:
@morrolinux yes, gpu_freq needs to be set on rpi3. You are running a GUI, aren't you? AFAIK it doesn't run well with GUI. I run my pi headlessly.

morrolinux, 2018-04-24:
Nope, I do never run GUI(s) :)

Miegl, 2018-04-24:
Well that's weird. Please get the latest firmware and kernel (Arch Linux Arm :) and try again. I won't try to make it work on older firmwares.

morrolinux, 2018-04-24:
Even weirdier: I've installed the latest arch arm on my Pi 1 and updated with pacman -Syu.
Added gpu_freq=250 in /boot/config.txt
The audio stutters for a few seconds and then the system freezes completely with audio looping the last fraction of second - just like it did on raspbian 
Ok, now I'm pretty lost

Miegl, 2018-04-24:
@morrolinux try unplugging the video output cable and try again with ssh

morrolinux, 2018-04-24:
Nope... I even tried disabling hdmi output with `/opt/vc/bin/tvservice -o` 
Stranger thing: I've noticed I cannot `cat /boot/config.txt` while the program is running as I get "Input/output error" but really, seems like all mmcblk device will have I/O errs when and only when the program is running

Miegl, 2018-04-24:
@morrolinux this should fix the I/O error: https://github.com/Miegl/PiFmAdv/commit/6730e23a2c60967c98ffc93f49b191a31923962a
Try it and let me know if I/O actually works now

morrolinux, 2018-04-24:
I'm afraid nothing changed:
https://imgur.com/a/8AsZ72c
https://imgur.com/a/UZPpLrZ

Miegl, 2018-04-24:
@morrolinux thanks im looking into it now

Miegl, 2018-04-24:
@morrolinux that might be actually a fault of new firmware... didn't get this before. We need open source firmware this is just... meh..

Miegl, 2018-04-24:
@morrolinux yeah its weird, idk what the heck are they changing in the firmware to break it everytime...

TheSneakySniper, 2018-04-25:
@Miegl 
Issue has been fixed for me, using the raspberry pi 3 B+
Here is my environment.
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

`Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux`

Works long range, just as before, and the audio does not get broken.

Good work!

TheSneakySniper, 2018-04-25:
CORRECTION: The battery bank I use does not impact range.

I have discovered that the LENGTH OF THE USB POWER WIRE is what causes the range to become limited.

JsBergbau, 2018-04-25:
@TheSneakySniper 
So the latest Raspian Image works with PiFMAdv flawlessly?

The shorther the cable the shorter the range or vice versa?

TheSneakySniper, 2018-04-26:
@JsBergbau The shorter the cable, the shorter the range.