miegl/PiFmAdv #28: termination issue with no audio

morrolinux, 2018-04-24:
What I'm doing: `sox someinput | pi_fm_adv ... --wait 0` 
I don't really understand why, but when sox playback finishes, the shell hangs indefinitely, there is no CTRL+C / CTRL+Z response, and killing -9 pi_fm_adv process seem to have no effect! the pid won't die, although the shell is working...

Miegl, 2018-04-24:
I think this is related to the i/o issue. So once that will be fixed (idk how and when) this should work.

morrolinux, 2018-04-24:
Damn! you're probably right.

Miegl, 2018-04-24:
With the current state of the raspberry pi firmware the only thing I can do now is trial and error. Don't know how long that will take though and theres a big possibility the firmware will break it anyway.

Niall7459, 2018-04-24:
I find that the previous way of modulating the wave before PLLC is more reliable. Although it would be great for the PLLC to work. And even then I think the DMA engine should be changed as that solved my problem for the GUI freezing

morrolinux, 2018-04-24:
@Niall7459 could you be a bit more specific about the "DMA engine change" stuff?

Miegl, 2018-04-24:
@Niall7459 yes it is. But it's also not very usable quality wise.
I've actually changed the DMA channel from 5 to 14 so I guess it can't be a problem now

Miegl, 2018-04-24:
So I just pushed https://github.com/Miegl/PiFmAdv/commit/c9a50abfe046d2fafab73623610c8785fc8ed002

Let me know if it works this time @Niall7459 @morrolinux (test on multiple pis with multiple systems please :)

Miegl, 2018-04-24:
The main point is to get it working on latest firmware, so if it doesn't work on older one it's not a problem

morrolinux, 2018-04-24:
Good News!
Termination works correctly on both my Pi1 and the Pi3 
As for the stability, I'll test further tomorrow

Niall7459, 2018-04-27:
@morrolinux The general crash time is 2-4 hours after the program has started, I cannot test at this moment in time.

closed 2018-05-12 by morrolinux

Niall7459, 2018-05-12:
@morrolinux @Miegl I can do some tests for like 24 hours etc in a couple of days. and il get back to you with the results.