miegl/PiFmAdv #33: Error: could not open input file udio

Brandon4466, 2018-06-18:
Hello! I've used this program for quite a while, however it was a few months ago that I stopped using it. I'm reinstalling it now and plan on using it!

I've got it installed, and I can start it and it says it's running (playing nothing), but once I try to use an audio file, or pipe audio to it, it errors and says the following:

```
Error: could not open input file udio
```

I've searched online but can't figure out what this means or how to fix it.

Any help would be appreciated!
Thank you!

Miegl, 2018-06-18:
Never encountered this problem. My guess is you don't have libsndfile installed.
In that case install it by running `sudo apt-get install libsndfile1-dev` (Raspbian)
`sudo pacman -S libsndfile` (Arch Linux)

or you didn't specify arguments properly.

GizmoTheGreen, 2018-06-19:
I've got this problem too. if I use --audio "filename.wav" it works
but if I try to use any piping, even copying/pasting th examples it gives said error.

`sox -t mp3 http://www.linuxvoice.com/episodes/lv_                        s02e01.mp3 -t wav -  | sudo ./pi_fm_adv -audio -
Carrier: 107.90 Mhz, VCO: 971.1 MHz, Multiplier: 50.578125, Divider: 9
Allocating physical memory: size = 4354048, mem_ref = 5, bus_addr = fe681000, vi                        rt_addr = 0x7678c000
Master PLLA Locked
PPM correction is 0.0000, divider is 2129.6052 (2129 + 2479*2^-12).
Error: could not open input file udio.
Terminating: cleanly deactivated the DMA engine and killed the carrier.
sox WARN wav: Length in output .wav header will be wrong since can't seek to fix                         it
`

similar problem with trying to do mic-input

Brandon4466, 2018-06-19:
@Miegl Haha, that's awesome! Thank you for figuring it out!

GizmoTheGreen, 2018-06-19:
aaah, see I was smart enough to fix that when playing locally (I prefer to use "long" arguments -- style if they exist) I didn't realise -a would parse as such without a space and though udio was a typo, as in "Can't find input audio file"

Miegl, 2018-06-19:
@GizmoTheGreen I don't know if pifmadv is quite ready to be used as an aux alternative - audio quality is decent but there's an issue where the program just freezes after some time and you have to restart it to make it work again.
Also make sure to use a very short antenna (to stay legal) :)

GizmoTheGreen, 2018-06-19:
@Miegl i can just repower the pi then :)

I just used a lower value for mpx to shorten distance, or isn this not the right way about it?

Miegl, 2018-06-19:
@GizmoTheGreen `--mpx` just sets volume of the audio (and stereo sub-carrier).
You can use the `--power` argument to set gpio drive strength.
```
0 = 2mA
1 = 4mA
2 = 6mA
3 = 8mA
4 = 10mA
5 = 12mA
6 = 14mA
7 = 16mA
```
So setting a lower `--power` will shorten the distance. By default it's set to 7. Sorry for not including this option in README.