miegl/PiFmAdv #37: Insert artist and song name in --rt  RDS radiotex [HOW-TO]

pilovis, 2018-09-10:
This is not an issue but an How To, sorry bau didn't found where to put this.

1) install ffmpeg:

download and compile ffmpeg from the source. 
https://www.ffmpeg.org/download.html
Unpack it.
tar -xvf ffmpeg-x.x.x.tar.bz2
then change to that directory and:
./configure
make
sudo make install

**NOTE: on PI2 the compile process take several housr!!**

2) my script to do the dirty job:

```
#!/bin/bash
cd /root
/bin/sleep 1
rm rds_ctl
/bin/sleep 1
mkfifo rds_ctl &
/bin/sleep 1
/usr/bin/sox --buffer 32768 --input-buffer 32768 -t mp3 lista.m3u -t wav -c 2 - | sudo ./PiFmAdv/src/pi_fm_adv --preemph eu --cutoff 15000 --dev 50 --ctl rds_ctl --freq 88.0 --audio - &
while :
do
ffprobe 2> /dev/null -show_format http://webradio.antennevorarlberg.at:80/lounge | grep TAG:StreamTitle= | cut -d '=' -f 2 | perl -ne 'print "PS 'RADIO-88'\nRT $_"' > /root/rds_ctl
/bin/sleep 10
done

```

Maurizio

pilovis, 2018-09-10:
note:
with the above script and PiFM trasmitter running, the total CPU usage on a Pi2 overclocked at 1000 Mhz, is about 40%
The total memory usage is about 140 Mbytes

pilovis, 2018-09-10:
Another bug:

Program quits after 3-4 hours, error: "Could not rewind in audio file. Terminating"

Proposal workaround: prebuffers the audio trough mpg123, then sox reads it as raw audio and outputs it as wav to PFR

sudo nano /root/inject-title.sh

```
#!/bin/bash
cd /root
/bin/sleep 1
rm rds_ctl
/bin/sleep 1
mkfifo rds_ctl &
/bin/sleep 1
mpg123 --buffer 2048 --no-icy-meta -q -s http://webradio.antennevorarlberg.at:80/lounge | /usr/bin/sox --buffer 32768 --input-buffer 32768 -t mp3 http://webradio.antennevorarlberg.at:80/lounge -t wav -c 2 - | sudo ./PiFmAdv/src/pi_fm_adv --preemph eu --cutoff 13000 --dev 50 --wait 0 --ctl rds_ctl --freq 88.0 --audio - &
while :
do
ffprobe 2> /dev/null -show_format http://webradio.antennevorarlberg.at:80/lounge | grep TAG:StreamTitle= | cut -d '=' -f 2 | perl -ne 'print "PS 'RADIO-88'\nRT $_"' > /root/rds_ctl
/bin/sleep 10
done

I'll give it a try and let you know if it solve the problem or not.

```

crisdash96, 2020-04-28:
the script works, but it doesn't update the song, only if I disconnect the signal can it be updated, I put the script accordingly, I even put the one on this page and it doesn't update either. I put any test for loop and it gave loop, so I don't know why the title of the song doesn't loop to update it automatically

crisdash96, 2020-09-28:
does no refresh te song, here are a one solution
edit a bin of a notepad

#!/bin/bash

while true
do

/usr/local/bin/ffprobe 2> /dev/null -show_format http://stream.zeno.fm/zt12d0kk5bruv | grep TAG:StreamTitle= | cut -d '=' -f 2 | /usr/bin/perl -ne 'print "RT $_"' | cat > rds_ctl 

sleep 10
exit 0

save te sh and rename notepad to inject-title.sh

so now, open a terminal
open te folder cd Desktop and cd PiFmAdv and cd src
mode supersu : sudo su
sudo /bin/sh /root/inject-title.sh &
cat >rds_ctl

its ok, now refresh the tittle and artist song, only this song artist is end, and pass the next song artist , the tittle song and artist apears

100% funcionally