miegl/PiFmAdv #39: Piping audio from USB microphone input has ticking sound in background

Nawor3565, 2018-10-02:
I just bought a USB microphone adapter, so I can use my phone to pipe audio directly into PiFmAdv. However, when piping audio, there is a ticking noise that can be heard around every second. I recorded a few seconds of this, can be heard [here](https://drive.google.com/file/d/1S_Kkq-spsUu52IhFmI6gSGMANEfKOkby/view?usp=sharing). This ticking sound can be heard whether my phone is actually playing audio or not, and even pressing the "mute microphone" button on my USB adapter doesn't stop the sound. I recorded audio with the adapter on my PC, and there isn't any ticking there, and it doesn't happen when transmitting a .wav file directly. I'm not sure what is causing this, but it's very audible even when playing music, so it's rather annoying.

Miegl, 2018-10-02:
Not sure what's causing this.
From my testing playing an audio file supported by pifmadv (libsndfile) is working just fine. But when piping the audio something goes wrong. The ticking noise appears and after a long period of time it even stops transmitting on the desired frequency completely.

Miegl, 2018-10-02:
Please try piping an audio stream with sox or ffmpeg to pifmadv just to confirm that this problem is not cause by the usb sound card.

Miegl, 2018-12-23:
Possibly related to https://github.com/Miegl/PiFmAdv/issues/40