miegl/PiFmAdv #41: Terminate issue

caraiz, 2018-10-17:
Hi,

im testing it with a Raspberry Pi 3 and 3B+, the radio works but when the program exits it brokes the network connections.
The wifi is still connected, but ping, ssh,... do not work anymore.

I have debugging a bit and it hangs at **terminate** function _"Set GPIOs to be an output (instead of ALT FUNC 0, which is the clock)."_

Trying with an older version, works fine at Pi 3, but at Pi3B+ occurs the same at code _"Switch EMMC over to PLLD"_

Thanks.

Miegl, 2018-10-17:
This is weird... Don't know if it's specific to pi 3 or if it's due to firmware updates. Can somebody test this on a raspi 2?

caraiz, 2018-10-19:
Hi, 

i found that the problem appears executing this line at terminate function:

gpio_reg[3] = (gpio_reg[3] & ~(7 << 12)) | (1 << 12); //GPIO34

Im using default gpio (4).

arhico, 2018-12-16:
Have the same problem on my zero W. Commented out gpio 34 release to get it to work ok.

Miegl, 2018-12-23:
Hm, weird. Works fine on my Pi 2. Maybe it has something to do with wifi?

Miegl, 2018-12-24:
GPIO 34 is in fact used by the SDIO interface, which is used for wifi. https://github.com/raspberrypi/linux/blob/rpi-4.14.y/arch/arm/boot/dts/bcm2710-rpi-3-b.dts#L48

Miegl, 2018-12-24:
I will update the README to warn about GPIO usage and improve the terminate function to set only the used GPIO.

closed 2018-12-24 by Miegl