miegl/PiFmAdv #48: RT A/B flag does not work

tanaka1892, 2019-01-11:
The RT A/B flag does not seem to work no matter how many times I try. I tried the following command and nothing seems to happen on the RT display in SDRsharp.

```echo -e "AB ON\nRT some text"```

Doing the commands one at a time makes no difference and RDS Spy does not show the bit changing.

Miegl, 2019-01-14:
Does the fifo pipo work at all? Try setting different PS to see if other commands work.

tanaka1892, 2019-01-15:
The control pipe works. I have a script that updates the RT with the currently playing song and does dynamic PS as well. The problem here is the RT display does not clear out when using the AB command.

My script pseudocode:
```
echo "RT Here's one song" > /tmp/rds
while songIsPlaying
  echo "PS Here's" > /tmp/rds
  sleep 4
  echo "PS one song" > /tmp/rds
  sleep 4
  ...
// when a new song comes up, update the RT.
echo "AB abState" > /tmp/rds
echo "RT Here's another song" > /tmp/rds
...
```

The script keeps track of the AB flag and flips between the ON and OFF states on every RT update.

tanaka1892, 2019-04-28:
I fixed the problem. The issue is the AB flag toggles the first bit in the 2nd block rather than the 5th bit. Here is the edit that needs to be done:

In src/rds.c, replace
`blocks[1] = 0x2000 | rds_params.tp << 10 | rds_params.pty << 5 | rds_params.ab | rt_state;`
with
`blocks[1] = 0x2000 | rds_params.tp << 10 | rds_params.pty << 5 | rds_params.ab << 4 | rt_state;`

This toggles bit 5 and now RT clears out as expected.

miegl, 2019-05-02:
Thank you! I guess I never tested the AB functionality :smile: 
Fixed in https://github.com/miegl/PiFmAdv/commit/fa28a04667490da1b04cc8e90ee4bbc01c10bde4

closed 2019-05-02 by miegl