miegl/PiFmAdv #49: Piping problem with cheap USB Sound card

StoneSet, 2019-01-28:
Hello,
I have some issue using a pipe with PiFmAdv, however I do not find this issue with PiFmRds.
When I execute the command: `sudo arecord -fS16_LE -r 44100 -Dplughw: 1.0 -c 2 - | sudo / home / pi / PiFmAdv / src / pi_fm_adv --freq 105.10 --audio -` no error but the program cuts off just after starting it.
[image]

Regards,
Thanks

(In pifmrds : `sudo arecord -fS16_LE -r 44100 -Dplughw:1,0 -c 2 -  | sudo /home/pi/PiFmRds/src/pi_fm_rds -freq 105.10 -audio -`)

Stinkers, 2019-01-29:
The way pipes are handled is not good. If there is an under-run the pipe needs to be opened again by pifm, but it can't because the pipe is opened outside of the main loop. I'm currently working on some modifications to improve this, but I haven't got much spare time to do it.

Niall7459, 2019-02-12:
I'm also having a similar problem with the latest version. This also seems to be happening using standard wav files, such as: `sudo ./pi_fm_adv --audio sound_22050.wav`

Niall7459, 2019-02-12:
It has a random chance of working. 6/10 times it doesn't. For audio streams its more like a 1/10 it will even begin to work. But at some point it get cuts off.