miegl/PiFmAdv #51: Modifying for use as an MPX generator

tanaka1892, 2019-05-05:
I want to use this as an FM MPX generator with a 192kHz sound card. I've removed most of the Raspberry Pi transmitter code so now I'm left with only the stuff needed to produce MPX. I'm able to upmix the output audio to 2 channels and write the samples to a .wav file but now I'm figuring out how to resample it to 192kHz and output the audio to a sound card without piping the output to aplay. Implementing a sound card interface with libao might be possible but I'm not really good at writing new code. :(

Anthony96922, 2019-07-14:
I have managed to make a fully functional MPX generator using this once I figured out how to use libsamplerate and libao. If you're interested: https://github.com/Anthony96922/mpxgen

closed 2019-07-14 by Anthony96922

miegl, 2019-07-14:
Fantastic work! Can I commit your changes some time in the future to further enhance this project? Thanks!

Anthony96922, 2019-07-14:
Go ahead! Feel free to add new improvements! :)