miegl/PiFmAdv #52: Is it possible to transmit different frequencies from different gpio at same time?

francisbritto, 2019-05-28:
Hi miegl,

In this project we can able to transmit one frequency from one GPIO at the instance.But my question here is ,any chance to transmit  different frequencies from different GPIO at same time.

In Read-me document mentioned like we can configure GPIO pins 4,20,32.If it is the case is it possible to transmit three different frequencies with the help of three GPIO pins at the same time?

Thanks in Adavnce.

miegl, 2019-05-28:
That might be possible - there are 3 general purpose clocks (GPCLK{0,1,2}) and 3 usable PLLs (PLLA, PLLC, PLLD). Although we can't really change PLLD's frequency.

francisbritto, 2019-05-28:
Hi miegl,

Thank you very much for your response.It will be very helpful to me.Could
you elaborate this thing?.It will be very helpful to us.

miegl, 2019-05-28:
Sure I could. A complete rewrite of pifmadv has been in my mind for a long time now.

francisbritto, 2019-05-29:
Hi miegl,

Thanks for your reply.I am new to this domain.So could tou explain me how actually Raspberry PI is working as FM transmitter.What are the process involved in that and Role of the each component involved in this process.Could you guide me?

Regarding my query ,if you release PiFmAdv latest version( with Multiple frequency transmission option)it will be very helpful for everyone.