miegl/PiFmAdv #55: Not work with the PI 4

cpnetserver, 2019-07-07:
Hy guys!
No problem with the Raspberry Pi 3B+...
The same O.P. [Debian 10], same configuration, but with the PI 4
PiFmAdv not work.

miegl, 2019-07-08:
If the Pi 4 is backwards compatible it should be only a matter of adding new offsets like in the case of Raspberry Pi 2.

cpnetserver, 2019-07-08:
How?
how? I'm not very experienced, thanks.

broken2048, 2019-07-17:
https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=244479&start=25

**GPIO clock source Oscillator runs at 54 MHz now and PLLD at 750 MHz.**

pi_fm_adv.c : 

```
#define XTAL_CLOCK                      19.2e6
#elif (RASPI) == 4                      // Raspberry Pi 4
#define PERIPH_VIRT_BASE                0xfe000000
#define PERIPH_PHYS_BASE                0x7e000000
#define DRAM_PHYS_BASE                  0xc0000000
#define MEM_FLAG                        0x04
#define XTAL_CLOCK                      54.0e6
#else
```
Now creates a carrier on the designated frequency, but no audio is transmitted. Still looking.

miegl, 2019-07-17:
Maybe a DMA problem? PiFmAdv uses channel 14, you could try using a different one.

cpnetserver, 2019-07-17:
I tried to apply the broken2048 code above, used different DMA channel, but still not working...
It returns ever:
Warning: Master PLLA NOT Locked

broken2048, 2019-07-17:
Thanks for the tip! Got it working using 
```
#define XTAL_CLOCK                      19.2e6
#define DMA_CHANNEL                     14
#elif (RASPI) == 4                      // Raspberry Pi 4
#define PERIPH_VIRT_BASE                0xfe000000
#define PERIPH_PHYS_BASE                0x7e000000
#define DRAM_PHYS_BASE                  0xc0000000
#define MEM_FLAG                        0x04
#define XTAL_CLOCK                      54.0e6
#define DMA_CHANNEL                     6
#else
```
Apparently DMA 11-14 won't work

> At least DMA channels 0-7 work as before, while channel 1 and 3 is used by the firmware and channel 7 is a LITE engine. Channel 8 should also work and is also a LITE engine, but has the same interrupt number as channel 7. Channel 9 and 10 are used by the firmware. There are new 40-bit DMA engines 11-14, which are unknown

Be aware the places in the code that reference XTAL_CLOCK are not shown above, as well as DMA_CHANNEL 14 must be removed and relocated to the #if block

cpnetserver, 2019-07-17:
Hi broken2048!
I tried but still not working... (sure my fault!)
Can You post the pi_fm_adv.c file with the changes, Thanks.

broken2048, 2019-07-17:
Make sure you change the RASPI var in the Makefile
Good luck.
[pi_fm_adv.zip](https://github.com/miegl/PiFmAdv/files/3403153/pi_fm_adv.zip)

cpnetserver, 2019-07-17:
broken2018 thanks for file but compiling it returns:
root@RP4:/home/pi/PiFmAdv/src # make
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 rds.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2 waveforms.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=2   -c -o pi_fm_adv.o pi_fm_adv.c
pi_fm_adv.c: In function ‘tx’:
pi_fm_adv.c:540:5: error: too many arguments to function ‘fm_mpx_open’
  if(fm_mpx_open(audio_file, DATA_SIZE, cutoff, preemphasis_cutoff, srate, nochan) < 0) return 1;
     ^~~~~~~~~~~
In file included from pi_fm_adv.c:26:
fm_mpx.h:8:12: note: declared here
 extern int fm_mpx_open(char *filename, size_t len, int cutoff_freq, int preemphasis_corner_freq);
            ^~~~~~~~~~~
make: *** [<builtin>: pi_fm_adv.o] Error 1

broken2048, 2019-07-17:
You will need to make a couple changes, I have custom code in there for manually setting the sample rate and channels for STDIN PCM streams without a header. Also, your Makefile must have -DRASPI=4. So, you will need the two files in this zip.
[makefileandfm_mpx.zip](https://github.com/miegl/PiFmAdv/files/3403287/makefileandfm_mpx.zip)

miegl, 2019-07-17:
Please don't share code snippets like that...
@broken2048 if you are willing to share your changes please make a proper pull requests, otherwise I will commit them myself. Thanks!

cpnetserver, 2019-07-17:
broken2048... nothing to do:
root@RP4:/home/pi/PiFmAdv/src # make
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=4 rds.c
rds.c: In function ‘set_rds_rt’:
rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.rt, rt, 64);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function ‘set_rds_ps’:
rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
     strncpy(rds_params.ps, ps, 8);
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=4 waveforms.c
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O3 -march=armv7-a -mtune=arm1176jzf-s -mfloat-abi=hard -mfpu=vfp -ffast-math -DRASPI=4   -c -o pi_fm_adv.o pi_fm_adv.c
pi_fm_adv.c: In function ‘tx’:
pi_fm_adv.c:540:5: error: too many arguments to function ‘fm_mpx_open’
  if(fm_mpx_open(audio_file, DATA_SIZE, cutoff, preemphasis_cutoff, srate, nochan) < 0) return 1;
     ^~~~~~~~~~~
In file included from pi_fm_adv.c:26:
fm_mpx.h:8:12: note: declared here
 extern int fm_mpx_open(char *filename, size_t len, int cutoff_freq, int preemphasis_corner_freq);
            ^~~~~~~~~~~
make: *** [<builtin>: pi_fm_adv.o] Error 1

broken2048, 2019-07-17:
@miegl , sorry I was in a rush and not thinking. You are correct.

gonzalezjo, 2019-07-29:
@broken2048 Do you think that you could make pull requests for your code? It would be really helpful as we're having trouble setting this up.

broken2048, 2019-07-31:
@gonzalezjo Pull request submitted

closed 2019-07-31 by miegl