miegl/PiFmAdv #58: Feed PiFmAdv with 192kHz MPX signal

ItsPops, 2019-07-26:
Hello, 
Is it possible to use something like Stereotool on a Windows computer to feed PiFmAdv with a real MPX signal, over a 192kHz soundcard or Ethernet with MicroMPX (skipping the MPX generation process on the Rasp) ?
This would allow to broadcast a processed signal with stereo & RDS directly in it, and use less CPU power I believe. 

ItsPops

miegl, 2019-07-26:
It's not currently implemented, but it is possible. Not sure about micrompx though as it is proprietary.
You could skip all the mpx generation and feed the mpx signal coming off I2S or a usb soundcard directly.
If you are having problems with cpu power, there is a bunch of things you can do. The code isn't really optimized and I'm using the `double` data type in a few places where there's practically no difference. Also you can play with compiler optimizations, the current level of GCC optimizations set in PiFmAdv's Makefile is `-O1` for debugging reasons. You can safely push it to `-O2` and try some other parameters. But don't compile it with `-O3`, otherwise the program will have unpredictable results and will malfunction at some point.

dabfree, 2019-07-27:
Hi,

I would suspect that even these proprietary things are simply encoding to
mono FLAC at 192 kHz sampling, and streamed in an ogg wrapper. ;)

R.

miegl, 2019-07-28:
Thanks for the insight! I have also thought about single frequency networks. Would a MPX signal in a ogg container with timestamps and NTP for synchronization be sufficient? DMA synchronization will be needed too of course. So if all the synchronization was done right, would the NTP error be small enough to allow the single frequency network to work?

dabfree, 2019-07-28:
Hi Joseph,

There are SFNs and SFNs! With FM, the idea is not to have too much coverage
overlapping between transmitters. The more accurate the synchronisation of
the audio, the pilot tone phase, the better the outcome will be. One
problem never goes away entirely with synchronous FM and that is multipath
distortion. Careful placement of the transmitters and choice of tx aerial
system can help.

I did wonder about this for PiFM as Logitek open-sourced the squeezebox
server (there are 3rd paty linux OSS clients) as the system attempts to to
play audio synchronously out of multiple clients across a LAN. It might be
worth looking at their strategy? TThe more accurate the sync, the better
the experience will be. I think NTP is a few mS to ~100 mS accurate, so
it's not ideal. But it is certainly better than nothing!

One could go to extremes and make a hat with a low-cost GPS module to
provide 1 PPS into a GPIO pin and use it as a clock source to ensure
(timestamped) audio frames are modulated at precisely the right time, and
to provide a trigger to ensure the pilot tone phases are 'all going the
right way'. As for carrier frequency, I think the Pi uses a 19.2 MHz
crystal. This is a fairly standard freq so there room for improving the
accuracy and stability of that, but the audio sync is likely to be more
important. PPS would be within a few uS between each one! Frequency
differences would result in hetrodyne, but it won't be so bad if the
modulation is in sync.

All of this said, there is also the FM capture effect to factor in. As long
as the overlapping signals are n dB different in strength, it will mask a
lot of the interference effects. in the overlapping areas where the
difference is not enough, the audio will suffer varying degrees of 'mush'.

A 'poor man' FM SFN using is well worth trying. Years ago I installed two
50 mW FM transmitters to cover a university campus. The audio was fed from
an analogue link, so it was synchronous and there was no attempt to
synchronise the pilot phase or carrier frequencies. There should therefore
have been a large, very noticable mush area but we drove back and forth
between them four a couple of hours and we couldn't hear where the overlap
was! Strange how the theory predicts a worse outcome than what is found in
practice,

From that experience I will say again: the accuracy of synchronising of the
modulation will dictate how well it works....

R.

Anthony96922, 2019-08-22:
I'm also looking for a way to feed 192kHz MPX audio into PiFmAdv. The MPX signal will be 1-channel raw PCM data sent over ethernet. Something like this:

On a computer
````
./mpxgen --out-file /tmp/mpx-out &
nc raspberrypi 1234 < /tmp/mpx-out
````
and on the Pi
````
./pi_fm_adv --audio /tmp/mpx-in &
nc -k -l -p 1234 > /tmp/mpx-in
````
The audio goes through filtering so the processing path would need to be bypassed.

HyperDX, 2020-04-22:
That's very good idea! With Stereotool I can easily change RDS/audio settings or insert artist/title in there. Also I can change stereo pilot and RDS level.

miegl, 2020-04-22:
@HyperDX you can do all of this with PiFmAdv too, no need for Stereotool ;) Inserting artist/title can be done very easily with a FIFO pipe (`--ctl`). I've even made a simple script for setting just that from an Icecast server, may post it somewhere. Changing the RDS level is also pretty easy, there's currently no option for it but if you head to https://github.com/miegl/PiFmAdv/blob/master/src/fm_mpx.c#L237, you can just multiply `rds_buffer[i]`. By default it's on a normal level (that's what most FM stations do anyways).

HyperDX, 2020-04-23:
I didn't know about that, thanks! For me Stereotool is/was the easiest way to provide artist and title info to transmitter. Could you share this script if you can? :D

miegl, 2020-04-23:
@HyperDX here it is: (didn't test it and made some adjustments, hopefully it will work :)

```
#!/bin/bash

fifo="rds_ctl";
[ -p $fifo ] || mkfifo $fifo;

old_content_file="old_rds.txt";
[ -f $old_content_file ] || touch $old_content_file;

echo "PS Radio" > $fifo;

while true
do
        content=$(curl -s <icecat server>/rds.xsl);
        content_old=$(cat "$old_content_file");

        prefix="* - "
        suffix=" - * - *"
        song=${content#$prefix}
        song2=$song
        show=${content%$suffix}
        name="$song / $show"

        if [ "$content" != "$content_old" ]; then
                echo "$content" > $old_content_file;
                if [ ${#name} -lt 49 ]; then
                        echo "RT $name / Radio" > $fifo;
                        echo "RadioText has been set to: $name / Radio";
                elif [ ${#name} -gt 62 ]; then
                        echo "RT $song" > $fifo;
                        echo "RadioText has been set to: $song";
                else
                        echo "RT $name" > $fifo;
                        echo "RadioText has been set to: $name";
                fi

                sleep 5;

                while [[ -n $song ]]; do                # loop until $song is empty
                        if [[ ${#song} -gt 8 ]]; then
                                b="${song:0:8}"         # get first 8 characters of $song
                                c="${b% *}"             # crop all from right incl. first whitespace
                        else
                                c="$song"
                        fi
                        echo "PS $c" > "$fifo"
                        d="${song#$c}"                  # crop $c left from string $song
                        song="${d# *}"                  # remove a leading whitespace
                        sleep 2.5
                done

                echo "PS  " > $fifo;
                sleep 2.5;

                while [[ -n $song2 ]]; do               # loop until $song2 is empty
                        if [[ ${#song2} -gt 8 ]]; then
                                b="${song2:0:8}"        # get first 8 characters of $song2
                                c="${b% *}"             # crop all from right incl. first whitespace
                        else
                                c="$song2"
                        fi
                        echo "PS $c" > "$fifo"
                        d="${song2#$c}"                 # crop $c left from string $song2
                        song2="${d# *}"                 # remove a leading whitespace
                        sleep 2.5
                done

                echo "PS Radio" > $fifo;
        fi

        echo "$content" > $old_content_file;
        sleep 2.5;
done
```

The script expects `rds.xsl` to contain pure text of format: `<show name> - <artist> - <song name>`

Samgames333, 2024-12-11:
https://github.com/miegl/PiFmAdv/blob/master/src/fm_mpx.c#L237 you can just multiply `rds_buffer[i]`...

I can't find 'rds_buffer'. How can I increase RDS injection in the latest version?

ryanginn, 2025-06-15:
The MPX input from a mono FLAC would be much easier. Any updates?