miegl/PiFmAdv #60: Super fast audio playback

KlonoaOfTheWind, 2019-08-01:
I'm pretty sure this is something that's on my end, but I'm just not sure what it could really be. I'm piping audio from mpg123 though sox then into pifmadv. Quality is good and everything but the music plays super fast at a high pitch.

My command to launch is as follows.
`sudo /usr/bin/mpg123 -4 -s -Z /home/pi/Music/* | sox -V -t raw -esi -b16 -c2 -r44100 - -t wav - | sudo /home/pi/PiFmAdv/src/pi_fm_adv --freq 95.5 --audio -`

I'm not sure if I did anything wrong, or if I'm forgetting something, but any insight is appreciated.

miegl, 2019-08-02:
Not sure what the problem is, but PiFmAdv is now capable of handling raw 16bit pcm streams (thanks to @broken2048 :). Just pipe the raw pcm stream to PiFmAdv and set --srate <sample rate> and --nochan <number of channels>.

KlonoaOfTheWind, 2019-08-02:
Oh, that's awesome! Didn't realize that it was merged 2 days ago. I did a complete reinstall of raspbian to clear alot of garbage, so I'll just go ahead and take this time to reinstall this and a few other things. 
Also thank you for the awesome project. I'll open another issue if I run into any trouble.

closed 2019-08-02 by KlonoaOfTheWind