miegl/PiFmAdv #68: Routing RPi2 audio output to --audio

mfsoftworks, 2019-12-31:
Is it possible to route a RPi 2 audio output to --audio for use with Mopidy or the like? Routing the audio output either directly to --audio or to some kind of temp file read by --audio would be good for curating a playlist remotely and broadcasting centrally

mfsoftworks, 2019-12-31:
It seems like `rec -d -t raw -t wav -r 44100 -c 2 - | sudo ./pi_fm_adv --audio - --ps NygmaRadio --freq 103.9 --tp 0 --power 7` might be the answer but I don't currently have any wire for the RPi to test so I'll check after I have some

miegl, 2019-12-31:
On Tue, Dec 31, 2019 at 09:26:31AM -0800, MF Softworks wrote:
> `rec -d -t raw -t wav -r 44100 -c 2 - | sudo ./pi_fm_adv --audio - --ps NygmaRadio --freq 103.9 --tp 0 --power 7`

Please note that the program service name (ps) can carry only 8 characters.

arranfletcher, 2020-01-01:
> On Tue, Dec 31, 2019 at 09:26:31AM -0800, MF Softworks wrote: `rec -d -t raw -t wav -r 44100 -c 2 - | sudo ./pi_fm_adv --audio - --ps NygmaRadio --freq 103.9 --tp 0 --power 7`
> Please note that the program service name (ps) can carry only 8 characters.

Oh thanks I missed that part, I'll do a test this afternoon

SimonSch, 2020-01-22:
Did you manage to get it to work?

miegl, 2020-05-04:
Closing, open a new issue if needed.

closed 2020-05-04 by miegl