miegl/PiFmAdv #69: Compiling Fix For Pi Zero W On Other Distro's

Dtpk, 2020-02-25:
Some distro's won't let you compile this even when the pi zero w supports hard floats. The simple fix is to edit the make file and change " ARCH_CFLAGS = -march=armv6 -O3 -mtune=arm1176jzf-s -mfloat-abi=hard "
too
" ARCH_CFLAGS = -march=armv6 -O3 -mtune=arm1176jzf-s -mfloat-abi=softfp "
or
" ARCH_CFLAGS = -march=armv6 -O3 -mtune=arm1176jzf-s -mfloat-abi=soft "
after that the program compiles fine seems to work ok. I can't guarantee how it preforms in long tests just figured i'd post this fix here as well as on the original project this is based off. Tested this fix on p4wn pi aola which is basically kali linux.

Dtpk, 2020-02-25:
Well never mind this fix isn't as simple as with pifm this works but after while now doesn't transmit.  If your desperate try my solution but it seems to stop working after sometime. No audio at all but still pretends to transmit. 
------------------------------------------------------------------------------------
Carrier: 86.90 Mhz, VCO: 955.9 MHz, Multiplier: 49.786458, Divider: 11
Allocating physical memory: size = 4354048, mem_ref = 5, bus_addr = 5e44e000, virt_addr = 0xb6733000
Master PLLA Locked
PPM correction is 0.0000, divider is 2096.2719 (2096 + 1113*2^-12).
RDS Options:
RDS: 1, PI: 1234, PS: "PiFmAdv", PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 86.9 MHz.

miegl, 2020-02-25:
That's probably due to -O3. Pifmadv doesn't work with level 3 optimizations (yet). -O2 is probably stable though. Hopefully I'll find time and improve the DMA bits so it doesn't crash.

On February 25, 2020 8:24:56 AM GMT+01:00, Dtpk <notifications@github.com> wrote:
> Well never mind this fix isn't as simple as with pifm this works but
> after while now doesn't transmit.  If your desperate try my solution
> but it seems to stop working after sometime. No audio at all [...]

Dtpk, 2020-02-27:
Changed the mfloat-abi back to the original "-mfload-abi=hard " . Then went tried to compile with -O2 optimizations it just fails at compiling unless I set the mfloat-abi to softfp. If I do that and leave the optimization too -O1 it will compile and work "correctly" but bluetooth cuts off if I ssh in via bluetooth and start transmitting how ever if I control the terminal via WiFi the sound transmits pretty ok with small hick-up's in the sound every now and then which  is probably "due to the low voltage detection the PLLC " so I suppose if I overlcock higher it would work fine on the Pi Zero W. But i'll keep a eye out on updates thanks for the comment and also just wanted to let you know I had a typo in the previous post the -O3 wasn't suppose to be there.