miegl/PiFmAdv #72: PiFmAdv on Raspberry pi 4

haiko1994, 2020-04-19:
Hello All! 

i'm new here and my english is also not the best but i try to do my best.
I hava bought an Raspberry pi 4 and want to install the PiFmAdv. But when i try to start the application i get a error message 'Warning: Master PLLPA Not locked' and i get no music on the selected Frequency. 

Is there someone who as a solution or the samen problem?

miegl, 2020-05-04:
The PLLA (Auxiliary PLL) is used for the camera interface, but changing its frequency can also collide with video from my experience. That's why I usually recommend running the Raspberry Pi headless (i.e. without graphical interface). Perhaps describe your setup a bit?

formerlyKnownAsLagzilla, 2020-06-09:
I'm having the same issue. I'm using image ```2020-05-27-raspios-buster-lite-armhf.img``` with no camera or desktop installed
**config.txt**
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

gpu_freq=250

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

dtoverlay=w1-gpio
start_x=0
```
**Output**
```
sudo ./pi_fm_adv stereo_44100.wav
Carrier: 87.60 Mhz, VCO: 1051.2 MHz, Multiplier: 19.466667, Divider: 12
Allocating physical memory: size = 4354048, mem_ref = 7, bus_addr = fe0b8000, virt_addr = 0xb67d6000
Warning: Master PLLA NOT Locked
PPM correction is 0.0000, divider is 2305.2632 (2305 + 1077*2^-12).
RDS Options:
RDS: 1, PI: 1234, PS: "PiFmAdv", PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 87.6 MHz.
```

miegl, 2020-06-09:
Can you try commenting `dtoverlay=vc4-fkms-v3d` ?

pwrobot, 2020-07-14:
I am having the same issue with my PI 4B saying "Warning: Master PLLA NOT Locked", have you been able to find a solution yet?

mobluse, 2020-12-17:
I have the same issue with "Warning: Master PLLA NOT Locked" on Raspberry Pi 4 B 4GB RAM and no sound or RDS (the same FM-radio in Android phone works fully with PiFmAdv on Raspberry Pi Zero WH):
```
$ sudo ./pi_fm_adv --audio sound_22050.wav
Carrier: 87.60 Mhz, VCO: 1051.2 MHz, Multiplier: 19.466667, Divider: 12
Allocating physical memory: size = 4354048, mem_ref = 4, bus_addr = feb9d000, virt_addr = 0xb6793000
Warning: Master PLLA NOT Locked
PPM correction is 0.0000, divider is 2305.2632 (2305 + 1077*2^-12).
Using audio file: sound_22050.wav
Input: 22050 Hz, upsampling factor: 10.34
1 channel, monophonic operation.
Created preemphasis with cutoff at 3185 Hz
Created low-pass FIR filter for audio channels, with cutoff at 15000 Hz
RDS Options:
RDS: 1, PI: 1234, PS: "PiFmAdv", PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 87.6 MHz.
```
I run pi_fm_adv after having rebooted after changing config.txt to this (no known config.txt works for me).
```
$ cat /boot/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

gpu_freq=250

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=2
#hdmi_mode=16

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#over_voltage=2
#arm_freq=1800

# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=off
#dtparam=i2s=on
dtparam=spi=off

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
gpu_mem=16

#arm_64bit=1

##turn on/ off bluetooth
dtoverlay=disable-bt

##turn on/off wifi
dtoverlay=disable-wifi
start_x=0
enable_uart=0
```
```
$ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 126.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03111
Serial		: 10000000095a18c0
Model		: Raspberry Pi 4 Model B Rev 1.1
```

miegl, 2022-04-15:
Fixed in https://github.com/miegl/PiFmAdv/commit/9f4cbc32d9361527d3d5f40c18a9f1b8150dcfa2
PiFmAdv now default to using PLLC, PLLA doesn't seem to work on Raspi 4 (it can be enabled with `--pll a`).

closed 2022-04-15 by miegl