miegl/PiFmAdv #74: Raspberry Pi4 - send own audio output (speaker output)

flopyro, 2020-05-21:
Hello, is it possible to broadcast my own Audio Output ?
So I have  Raspi4 which is connected to a Projector. 
I project the a Movie on the neighbours house, and also the other neighbours are watching, and i tought that i can send the sound of the movie thru FM Radio, so the other neighbours can also hear the sound, not only the Image.
Do you know how i could send the Audio from the movie ? I'm usind Kodi as a media player.

Thank you

flopyro, 2020-05-28:
ok, i cannot get it to work on PI4. works with other versions PiFmRSD, or with FM_Transmitter, but with this one is not :(
regardin the answer you gave laste year:

pacmd load-module module-pipe-sink file=/tmp/pulse.fifo
pacmd list-sinks  # search your default sink
pacmd load-module module-combine-sink slaves=yoursink,fifo_output
pacmd set-default-sink combined
And then run pi_fm_adv --audio - < /tmp/pulse.fifo

Let me know if this works.

there is something missing here: 
pacmd load-module module-combine-sink slaves=yoursink,fifo_output

another argument or something.

Anyway, thank you for your work. !!!

flopyro, 2020-05-28:
Now i get an errror when trying to start the station:
pi@raspberrypi:~/radio/PiFmAdv/PiFmAdv $ pacmd load-module module-pipe-sink file=/tmp/pulse.fifo                                                                pi@raspberrypi:~/radio/PiFmAdv/PiFmAdv $ pacmd load-module module-combine-sink slaves=alsa_output.platform-soc_audio.analog-mono,fifo_output sink_name=combined sink_properties=device.description=CombinedSinkt
pi@raspberrypi:~/radio/PiFmAdv/PiFmAdv $ cd src                                 pi@raspberrypi:~/radio/PiFmAdv/PiFmAdv/src $ sudo ./pi_fm_adv --audio - < /tmp/pulse.fifo                                                                       Carrier: 87.60 Mhz, VCO: 1051.2 MHz, Multiplier: 54.750000, Divider: 12
Allocating physical memory: size = 4354048, mem_ref = 8, bus_addr = fdfa7000, virt_addr = 0xb679b000
Warning: Master PLLA NOT Locked
PPM correction is 0.0000, divider is 2305.2632 (2305 + 1077*2^-12).
Error: could not open stdin for audio input.
Terminating: cleanly deactivated the DMA engine and killed the carrier.
pi@raspberrypi:~/radio/PiFmAdv/PiFmAdv/src $

dabfree, 2020-05-30:
It has been a long day, so I may well be wrong, but it doesn't look like
you are piping anything to the input, but you are calling stdin.

Maybe you could create some ALSA virtual sound devices, one playing back
the movie soundtrack and the other used as a capture device...but I am not
so sure if PiFmAdv works on a Pi4 though.

R.

miegl, 2022-04-16:
Easiest solution for piping Pulseaudio is using `parec`.

closed 2022-04-16 by miegl