miegl/PiFmAdv #76: Pi4 Help!

Marrley, 2020-06-24:
### Hello!

I've tried/researched many different types/variants of Pi_FM or RPiTx, but they all seem to yield the same results/problems when it comes to the Raspberry Pi 4.

**[Original Pi_FM](https://github.com/rm-hull/pifm):** No errors but no TX Output

**[PiFmAdv](https://github.com/miegl/PiFmAdv/):** No errors but no TX Output (same as Pi_FM)

**[PiFmRds](https://github.com/ChristopheJacquet/PiFmRds/):** No errors but no TX Output (same as Pi_FM)

**[PiFmRds (modded)](https://github.com/markondej/PiFmRds/):** Very low volume output and lots of buzzing/cracking noises

**[RPiTx](https://github.com/F5OEO/rpitx/):** Not working (as of reports from the [/issues](https://github.com/F5OEO/rpitx/issues/))

I can fairly say I have almost searched the entire internet forum system...

**_Two Questions:_**

_**1:** Does anyone know of/have a modified version of PiFmAdv or RPiTx that will work on the Raspberry Pi 4?_

_**2:** Is it possible to use the Raspberry Pi 4 as a standalone RDS transmitter alongside and external FM Transmitter that does not support RDS?.. or will the frequencies just cancel each other out?_

Thx! 
-Marley

miegl, 2020-06-24:
Describe your setup a bit more. Are you using graphical environment or are you using the video driver? Did you set the gpufreq value in your config?

I find it weird that the modified version of PiFmRds which uses GPIO21 works. The alt function is set to GPCLK0 as opposed to GPCLK1 in PiFmAdv. I will take a look on the device tree files of pi 4 to see what might be causing the issue.

2) Currently there's no way of feeding pifmadv to an external FM transmitter, but if you manage to change the program behaviour to output a 192khz sound and your FM transmitter doesn't feature a low pass filter it could possibly work.

On June 25, 2020 12:31:34 AM GMT+02:00, Marrley <notifications@github.com> wrote:
> ### Hello!
>
> I've tried/researched many different types/variants of Pi_FM or RPiTx,
> but they all seem to yield the same results/problems when it comes to
> the Raspberry Pi 4.
>
> **[Original [...]

Marrley, 2020-06-24:
###  **thx for your prompt reply!**

I know absolutely NOTHING about whatever code you used to make your application... but seeing as PiFmRds And Your PiFmAdv use roughly the same base code, I was wondering if its possible, to take whatever is making PiFmRds work and just _**magically pop it in**_ :/

_Referring to your:_
**I find it weird that the modified version of PiFmRds which uses GPIO21 works. The alt function is set to GPCLK0 as opposed to GPCLK1 in PiFmAdv.**

Marrley, 2020-06-25:
@miegl **_also note_** when using PiFmRds(modded) sound is outputted at very low volume and quality. Alongside that there is also cracking noises + bee like noises.

miegl, 2020-06-25:
Please respond to the first two questions. I'm also pretty sure there are guys who had success with a pi 4, so maybe try to elaborate how your setup could differ from theirs?

Marrley, 2020-06-25:
Oh sorry I guess I skimmed past your first two questions

Here is my setup:

Model: Raspberry Pi 4 - Model B
PSU: default raspberry pi power supply from a kit
FAN: no fan

I use my pi headless with Putty, however Raspbian was installed using NOOBS and a display

Here are the specs directly from: `cat /proc/cpuinfo`

processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : b03111
Serial          : 100000001cca5704
Model           : Raspberry Pi 4 Model B Rev 1.1

OS:

Debian Version: 10.4

Here is directly from: `cat /etc/os-release`

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

Currently Running apache2

miegl, 2020-06-25:
Thanks, did you set your /boot/config.txt correctly as stated in README?
If it still doesn't work after that, try disabling your display manager (so the graphic environment doesn't come up).

Marrley, 2020-06-25:
i dont want to reset the os completely because i would have no way of re-installing...
and i cant open the boot sd card on my pc... so is there another way i can edit config.txt from ssh or ftp?

Marrley, 2020-06-25:
Nope Tried the gpu_freq 250 then reboot thing it didnt work

Marrley, 2020-06-25:
also if i cant get this to work might you have a way to get rid of the buzzing and cracking sounds in PiFmRds (modded)

FYI before i transmit i use: `echo "performance"| sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor`
 It helps get rid of some of the pulsing sounds

miegl, 2020-06-26:
Try running `sudo systemctl disable lightdm`
This should disable starting the X server at boot.

Marrley, 2020-06-27:
nope didn't work either

Ben1701A, 2021-04-08:
I got a copy of rpitx to run on my Raspberry Pi 400 before. No GPU safe clocking, no display, just me with an iPad and VNC Viewer. 
However, since I've kept it plugged in to my monitor, running rpitx crashes the GPU. Everything else runs normally--it's only the GPU that goes.

TheLiuzhu, 2021-05-25:
The RPItx works on my's 4B too, but I can't stop it from transmitting one file continually in a loop (not a list),which it's hard to use for my playlist service🤣🤣😓.

miegl, 2022-04-15:
Support for Raspi 4 was fixed in https://github.com/miegl/PiFmAdv/commit/9f4cbc32d9361527d3d5f40c18a9f1b8150dcfa2

closed 2022-04-15 by miegl