miegl/PiFmAdv #77: gpio no transmit on raspberry pi 3

crisdash96, 2020-08-08:
l have a problem, l conect the antenna to the pines 3, 20, 32, only 32, is transmit, beacuse transmit very little, others pines whit antenna no transmit, I don't know why that happens.

crisdash96, 2020-08-13:
Im buyed other raspberry (3b+) and works. Is for are a new raspberry. I have no solution for the raspberry 3b normal. Close this

miegl, 2020-08-14:
Reopening as this should probably get fixed.

crisdash96, 2020-08-17:
can you enable the gpio 21 to stream? It is the only one that works for me to transmit in the normal raspberry 3b

crisdash96, 2020-09-23:
im sold the other raspberry with failed pins 8 and 20, beacuse no have solution.

miegl, 2022-04-16:
Current version allows transmitting on all available GPCLK pins by setting `--gpio` and `--gpclk`.

closed 2022-04-16 by miegl

crisdash96, 2022-04-17:
The raspberry of months past have problems I’m devolved to the store, sorry raspberry it’s no my option for 4 months have a real fm transmitter 15 watts no problems and rds works witch 192khz sound card.