miegl/PiFmAdv #81: Playing stereo_44100.wav too slow and without RDS

andimik, 2020-12-23:
I am using a RPI3B and could build PiFmAdv without any problem.

But when I want to test it with the `stereo_44100.wav` I hear a male voice but this is very very slow. I can even not understand what the voice is saying.

```
$ sudo ./pi_fm_adv -a stereo_44100.wav -f 87.9
Carrier: 87.90 Mhz, VCO: 966.9 MHz, Multiplier: 50.359375, Divider: 11
Allocating physical memory: size = 4354048, mem_ref = 8, bus_addr = fe40e000, virt_addr = 0x767ca000
Master PLLA Locked
PPM correction is 0.0000, divider is 2120.3947 (2120 + 1616*2^-12).
Using audio file: stereo_44100.wav
Input: 44100 Hz, upsampling factor: 5.17
2 channels, generating stereo multiplex.
Created preemphasis with cutoff at 3185 Hz
Created low-pass FIR filter for audio channels, with cutoff at 15000 Hz
RDS Options:
RDS: 1, PI: 1234, PS: "PiFmAdv", PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 87.9 MHz.
^CTerminating: cleanly deactivated the DMA engine and killed the carrier.
```

I've added `gpu_freq=250` to `/boot/config.txt`

evertverduin, 2021-01-01:
There is no support, at least, no one responds. 

My solution was to erase raspbian os! 
install Raspbian Jesse lite, and install gpu_freq=250 in config.txt

Symptoms are:
Baseband signal is at half of it's sampling rate. Audio to slow, pilot on 9.5 khz and rds at rds/2.

Solution -> install Raspbian Jesse. with gpu_freq=250 in config.txt

BR

Evert

andimik, 2021-01-20:
I solved it by using https://github.com/mundeeplamport/PiFM instead.

No need to reinstall OS.