miegl/PiFmAdv #82: RDS errors during reception

HyperDX, 2021-01-17:
I connected RDS Spy to my receiver and tuned to my radio station and the BER (Bit Error Rate) independently from the signal level (medium, strong) sometimes shows few percent error rate. Also the Group Analyser shows missing groups (Screenshoot). When I'm tuned to other radio stations "BER" is always at 0% and there are not any missing groups in the Group Analyser. This could affect slower loading of RDS PS, RDS RT etc. I checked also PiFmRDS and this problem doesn't exist there. But this software is really simple and does not have as many options as PiFmAdv.
[image]

HyperDX, 2021-04-24:
I also noticed that this problem doesn't occur when the CPU usage is quite high.

miegl, 2022-04-16:
Please check if the problem is still present in current version. Thanks.

HyperDX, 2022-04-19:
Running new version with default settings just bricks my RPi 3B+. So I can only unplug and plug again power cord to make it alive.

miegl, 2022-04-21:
Please try running with `--pll a`.

HyperDX, 2022-04-21:
Now it's working fine with --pll a option. Thanks!

Sadly RDS problem described in a 1st comment is still present. :(

HatKidTheHatLoli, 2022-08-11:
Can confirm that Ctrl+C does not kill the carrier. Pi 2 B, fairly fresh install.