miegl/PiFmAdv #84: Station dies after about an hour

cjuniorfox, 2021-08-19:
I'm piping from arecord using snd_aloop device.
The oscillation just dies after about one hour or two.
The pi_fm_adv does not report any issue. The process keeps running like there's nothing wrong happened.

May be related with #40 but I really don't know.

I'm running raspibian jessy on Raspberry Pi 3B

Command follows

`arecord -fS16_LE -r 44100 -Dplughw:0,0 -c 2 - --quiet | pi_fm_adv --wait 0 --freq 91.5 --audio - --ctl /tmp/rds_ctl --mpx 40 --preemph us`

dabfree, 2021-08-21:
Hi,

I can't quite believe that I am about to post the same explanation I gave to another repo earlier on today! :-D

The problem is almost certainly arecord rolling the output every 2GB - a 32-bit limitation. I ran into a similar problem about eight years ago when using arecord to open an ALSA input device!

The workaround I found was to use ecasound to open the ALSA device instead and the problem went away.

Hope this works for you.

R

cjuniorfox, 2021-08-21:
Thank you for your response.
I installed ecasound and I successfully able to capture my alsa device to one wav file, but how do I write ecasound to stdout to pipe pi_fm_adv just like I do with arecord?

cjuniorfox, 2021-08-25:
I believe the problem does not rely on aplay's buffer
I tested several ways to feed the pifmadv. I tried pipping ffmpeg to stdin, with wav files, using named fifos. All with no success.

Additionaly, I tested the [fm_transmitter](https://github.com/AshrafAkon/fm_transmitter.git) (mono, 22050hz) with no problems at all.
Unfortunely, with PiFmAdv, the oscillation is hardly sustained for more than some couple hours.