miegl/PiFmAdv #95: Latest version will not run on Raspberry Pi4 even with pi user created

shades-aus, 2022-07-19:
Unsure what to say here. I have tried to get your program working on a fresh install of Raspian and followed the guide step by step and I am unable to get the program working from the terminal.
Followed the ReadMe file step by step to download from git and I cannot get it to work.

Looking inside setup-pi4.sh

#!/bin/bash

# This is a setup script for the Pi FM RDS Transmitter project streamlined by Mundeep Lamport
clear
echo "Welcome to the PiFM installer. This script will help you get up and running."
echo "This script is designed specifically for the Raspberry Pi 4."
echo "For more information, visit https://github.com/mundeeplamport"
echo "WARNING! Your system will reboot once completed."
sleep 10
clear
echo "Updating repository" ; sleep 1
sudo apt-get update
echo " "
echo "Installing required tools" ; sleep 1
sudo apt-get install libsndfile1-dev zenity git -y
clear
echo "Changing directory --> src" ; sleep 1
cd /home/pi/PiFM/src
clear
echo "Cleaning up" ; sleep 1
make clean
echo " "
echo "Compiling" ; sleep 1
make GPIO21=1
clear

Running "make" as defined by your script from inside src gives the following.

pi@ads-pi4:~/PiFM/src $ sudo make GPIO21=1
[sudo] password for pi:
gcc -Wall -Wno-multichar -std=gnu99 -c -g -O1 rds_wav.c
rds_wav.c: In function ‘main’:
rds_wav.c:53:8: error: too few arguments to function ‘fm_mpx_open’
   53 |     if(fm_mpx_open(in_file, LENGTH) != 0) {
      |        ^~~~~~~~~~~
In file included from rds_wav.c:32:
fm_mpx.h:8:12: note: declared here
    8 | extern int fm_mpx_open(char *filename, size_t len, int cutoff_freq, int preemphasis_corner_freq, int srate, int nochan);
      |            ^~~~~~~~~~~
rds_wav.c:81:32: warning: passing argument 1 of ‘fm_mpx_get_samples’ from incompatible pointer type [-Wincompatible-pointer-types]
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |                                ^~~~~~~~~~
      |                                |
      |                                float *
In file included from rds_wav.c:32:
fm_mpx.h:9:39: note: expected ‘double *’ but argument is of type ‘float *’
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, double *rds_buffer, float mpx, int rds, int wait);
      |                               ~~~~~~~~^~~~~~~~~~
rds_wav.c:81:13: error: too few arguments to function ‘fm_mpx_get_samples’
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |             ^~~~~~~~~~~~~~~~~~
In file included from rds_wav.c:32:
fm_mpx.h:9:12: note: declared here
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, double *rds_buffer, float mpx, int rds, int wait);
      |            ^~~~~~~~~~~~~~~~~~
make: *** [Makefile:51: rds_wav.o] Error 1

---------

After trying to run pifm-basic, I get the return after filling in all the fields

Completed. Starting service Pi FM.
sudo: ./pifm: command not found
pi@ads-pi4:~/PiFM $

Looking in pifm-basic I find the below

cd /home/pi/PiFM/src
sudo ./pifm --freq $frequency --ps $ps --rt "$rt" --audio $audio --pi $pi --pty $pty --mpx $mpx --power $power --preemph $pre

listing the files found in src, there is no PiFM file

pi@ads-pi4:~/PiFM/src $ ls -la
total 6556
drwxr-xr-x 3 ads  ads     4096 Jul 20 07:38 .
drwxr-xr-x 6 ads  ads     4096 Jul 20 07:20 ..
-rw-r--r-- 1 ads  ads     3334 Jul 20 07:20 control_pipe.c
-rw-r--r-- 1 ads  ads      460 Jul 20 07:20 control_pipe.h
-rw-r--r-- 1 ads  ads     7219 Jul 20 07:20 fm_mpx.c
-rw-r--r-- 1 ads  ads      386 Jul 20 07:20 fm_mpx.h
-rw-r--r-- 1 ads  ads      504 Jul 20 07:20 generate_pulses.py
-rw-r--r-- 1 ads  ads     1709 Jul 20 07:20 generate_waveforms.py
-rw-r--r-- 1 ads  ads     6929 Jul 20 07:20 mailbox.c
-rw-r--r-- 1 ads  ads     2381 Jul 20 07:20 mailbox.h
-rw-r--r-- 1 ads  ads     1674 Jul 20 07:20 Makefile
-rw-r--r-- 1 ads  ads   220544 Jul 20 07:20 noise_22050.wav
drwxr-xr-x 2 ads  ads     4096 Jul 20 07:20 pi4
-rw-r--r-- 1 ads  ads    28018 Jul 20 07:20 pi_fm_adv.c
-rw-r--r-- 1 ads  ads      196 Jul 20 07:20 PiFM.desktop
-rw-r--r-- 1 ads  ads    18303 Jul 20 07:20 PiFM.png
-rw-r--r-- 1 ads  ads        1 Jul 20 07:20 placeholder
-rw-r--r-- 1 ads  ads  6840044 Jul 20 07:20 pulses.wav
-rw-r--r-- 1 ads  ads     7908 Jul 20 07:20 rds.c
-rw-r--r-- 1 ads  ads      576 Jul 20 07:20 rds.h
-rw-r--r-- 1 root root   21520 Jul 20 07:38 rds.o
-rw-r--r-- 1 ads  ads     2669 Jul 20 07:20 rds_wav.c
-rw-r--r-- 1 ads  ads   188974 Jul 20 07:20 sound_22050.wav
-rw-r--r-- 1 ads  ads  1181958 Jul 20 07:20 sound.wav
-rw-r--r-- 1 ads  ads   352878 Jul 20 07:20 stereo_44100.wav
-rw-r--r-- 1 ads  ads    10206 Jul 20 07:20 waveforms.c
-rw-r--r-- 1 ads  ads      107 Jul 20 07:20 waveforms.h
-rw-r--r-- 1 root root    4584 Jul 20 07:38 waveforms.o

Raspberry Pi OS x64
Linux ads-pi4 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
Could your readme please be checked and updated with troubleshooting advice?
Any help getting this to work would be great.

shades-aus, 2022-07-19:
--Also in PiFmAdv--

pi@ads-pi4:~/PiFmAdv/src $ sudo make clean
rm -f *.o
pi@ads-pi4:~/PiFmAdv/src $ sudo make
gcc -Wall -Wno-multichar -std=gnu99 -c -O2 rds.c
rds.c: In function ‘set_rds_rt’:
rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
  240 |     strncpy(rds_params.rt, rt, 64);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
rds.c: In function ‘set_rds_ps’:
rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
  247 |     strncpy(rds_params.ps, ps, 8);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
gcc -Wall -Wno-multichar -std=gnu99 -c -O2 waveforms.c
gcc -Wall -Wno-multichar -std=gnu99 -c -O2 rds_wav.c
rds_wav.c: In function ‘main’:
rds_wav.c:53:8: error: too few arguments to function ‘fm_mpx_open’
   53 |     if(fm_mpx_open(in_file, LENGTH) != 0) {
      |        ^~~~~~~~~~~
In file included from rds_wav.c:32:
fm_mpx.h:8:12: note: declared here
    8 | extern int fm_mpx_open(char *filename, size_t buf_len, int cutoff_freq, int preemphasis_corner_freq, int srate, int nochan);
      |            ^~~~~~~~~~~
rds_wav.c:81:32: warning: passing argument 1 of ‘fm_mpx_get_samples’ from incompatible pointer type [-Wincompatible-pointer-types]
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |                                ^~~~~~~~~~
      |                                |
      |                                float *
In file included from rds_wav.c:32:
fm_mpx.h:9:39: note: expected ‘double *’ but argument is of type ‘float *’
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, int mpx_buffer_len, float mpx, int rds, int wait);
      |                               ~~~~~~~~^~~~~~~~~~
rds_wav.c:81:13: error: too few arguments to function ‘fm_mpx_get_samples’
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |             ^~~~~~~~~~~~~~~~~~
In file included from rds_wav.c:32:
fm_mpx.h:9:12: note: declared here
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, int mpx_buffer_len, float mpx, int rds, int wait);
      |            ^~~~~~~~~~~~~~~~~~
make: *** [Makefile:51: rds_wav.o] Error 1
pi@ads-pi4:~/PiFmAdv/src $

frnck37, 2024-04-06:
same pb here