miegl/PiFmAdv #96: Carrier signal stops being generated; program continues to run.

HatKidTheHatLoli, 2022-08-11:
gpu_freq=250 is in the boot config file before you ask. 
Often, it will stop producing the signal altogether. It appears to still be "working", like it's still trying to modulate. But of course nothing is being transmitted.
Also some strange behavior, if I close the program (Ctrl+C) or if the pipe is broken, the carrier signal does not stop being generated, and it will continue to produce the signal. If I run the command again, it'll keep generating the signal without cutting out, but will eventually hit a point where it says "waiting" before terminating itself (It doesn't kill the carrier frequency at this point either.) I assume the waiting thing is part of the buffer underflow/overflow issue mentioned in other issues.

PetitNigaud, 2022-08-27:
I can confirm. Before I launch the broadcast the receiver radio plays just noise, then with PiFmAdv it plays the desired sound, but when I stop its execution with CTRL+C the receiver radio is just silent. Only when I shut down the RPi it plays noise again.

ryanginn, 2022-08-28:
I can confirm as well.

The carrier is still active, however no sound is produced.

It seems that I have to reboot the Pi or kill the task to stop it.

craftxbox, 2023-10-07:
Make sure you have no GPIO modules loaded in the kernel. For me, this kept happening until I removed the one wire module: `sudo rmmod w1-gpio`