miegl/PiFmAdv #98: raspberry pi 4 no transmitting

jezdimsivefiatu, 2022-11-05:
nothing came out or noise on the raspberry pi 4.. Don't know how to fix it?

jezdimsivefiatu, 2022-11-07:
Ahoj, napišu to i český když vidím že jsi čech :D mám raspberry pi 4 8gb ram. A mám menší problem, nevysila vubec nic antenu všechno mám.. S PiFmRds nemám problém.. Až na to že šumi, a chybi mi par funkci co jsou tu, nějake rady ?

miegl, 2022-11-07:
Co mas za system a architekturu? PiFmAdv zatim funguje jenom na 32bit rpi-linux jadre, na 64bit aarch64 mainline jadre jsem netestoval a pravdepodobne nefunguje.
Hod sem taky vystup programu.

jezdimsivefiatu, 2022-11-07:
ahoj, mam normalne 32bit Raspian nejnovejsi verzi, a pouzival jsem i verzi 2021 :/

root@inovation:/home/pi/PiFmAdv/src# ./pi_fm_adv --freq 91.15--audio sound.wav
Carrier: 91.15 Mhz, VCO: 911.5 MHz, Multiplier: 47.473958, Divider: 10
Allocating physical memory: size = 4354048, mem_ref = 8, bus_addr = fe6cf000, virt_addr = 0xb665c000
Master PLLC Locked
Outputting signal on GPIO 4.
PPM correction is 0.0000, divider is 1096.4912 (1096 + 2012*2^-12).
RDS Options:
RDS: 1, PI: 1234, PS: "PiFmAdv", PTY: 15
RT: "PiFmAdv: Advanced FM transmitter for the Raspberry Pi"
Starting to transmit on 91.1 MHz.

jezdimsivefiatu, 2022-11-07:
s deb 11 :/ nejake rady ?

jezdimsivefiatu, 2022-11-07:
jestli mas discord, muzes mi napsat // matyas#7143

miegl, 2022-11-09:
`gpu_freq=250` v `/boot/config.txt` mas?

jezdimsivefiatu, 2022-11-09:
ano, a zkousel jsem i vypnout lighttdm

jezdimsivefiatu, 2022-11-09:
zkousel jsem prave ted schvalne cisty raspbian bez GUI, a taky nefunguje

jezdimsivefiatu, 2022-11-12:
napada te neco ?

Cheerstoast, 2023-06-24:
I use 64 bit, and I successfully get a ~20 meter radius on 88.0khz and a 10cm jumper wire, with software: [markondej/fm_transmitter](https://github.com/markondej/fm_transmitter)