miegl/PiFmAdv #99: Error while making

Cheerstoast, 2022-11-28:
The following 4 commands are executed, and this is the console output.
git clone https://github.com/Miegl/PiFmAdv.git
cd PiFmAdv/src
make clean
make

    Cloning into 'PiFmAdv'...
    remote: Enumerating objects: 297, done.
    remote: Counting objects: 100% (36/36), done.
    remote: Compressing objects: 100% (17/17), done.
    remote: Total 297 (delta 20), reused 26 (delta 19), pack-reused 261
    Receiving objects: 100% (297/297), 2.76 MiB | 7.82 MiB/s, done.
    Resolving deltas: 100% (200/200), done.
    rm -f *.o
    gcc -Wall -Wno-multichar -std=gnu99 -c -O2 rds.c
    rds.c: In function ‘set_rds_rt’:
    rds.c:240:5: warning: ‘strncpy’ specified bound 64 equals destination size [-Wstringop-truncation]
      240 |     strncpy(rds_params.rt, rt, 64);
          |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    rds.c: In function ‘set_rds_ps’:
    rds.c:247:5: warning: ‘strncpy’ specified bound 8 equals destination size [-Wstringop-truncation]
      247 |     strncpy(rds_params.ps, ps, 8);
          |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    gcc -Wall -Wno-multichar -std=gnu99 -c -O2 waveforms.c
    gcc -Wall -Wno-multichar -std=gnu99 -c -O2 rds_wav.c
    rds_wav.c:28:10: fatal error: sndfile.h: No such file or directory
       28 | #include <sndfile.h>
          |          ^~~~~~~~~~~
    compilation terminated.
    make: *** [Makefile:51: rds_wav.o] Error 1

jezdimsivefiatu, 2022-12-17:
do you have 32bit?

Cheerstoast, 2022-12-21:
No. I guess this is the problem :|

Prograb, 2023-04-09:
Wait so I need to run my Pi 4 in 32 bit mode ?

Cheerstoast, 2023-04-13:
@Prograb I use 64 bit, and I successfully get a ~20 meter radius on 88.0khz and a 10cm jumper wire, with software: https://github.com/markondej/fm_transmitter

closed 2023-04-13 by Cheerstoast