miegl/PiFmAdv #106: Does PiFM will work on Raspberry Pi 5 ?

joevin-slq, 2023-11-13:
Hello !

Just small question: is PiFM will work with the latest Raspberry Pi 5 ?
we can assume yes, because there is always the GPCLK output pin

has anyone tried it?
Thanks!

ryangontv, 2023-11-15:
I would assume so, considering it partially works with the Pi 4.

Must give it a shot when I order a Pi 5 in the near future.

KubaPro010, 2023-11-17:
Probably no, as the gpio pins won't be driven by the cpu but the RP1 and it have diffrent addresses to stuff and so it won't work

ryangontv, 2023-11-20:
> Probably no, as the gpio pins won't be driven by the cpu but the RP1 and it have diffrent addresses to stuff and so it won't work (you'll need to edit the code)

Interesting, Will have to give it a shot, heard big things about the RP1 chip.

Ryan