miegl/PiFmAdv #107: Multiple PS / Switching between 2 PS values

ValentinVV1, 2023-11-16:
Hi!

I hope that somebody still reads these.

I would like to know if it is possible to set 2 PS values to change automatically every few seconds and how to do it.

Someone else made an issue about this and said that he got it eventualy but didn't say how.

Thank you!

KubaPro010, 2023-11-17:
I made this python script a while back, i don't know if it works but you can test it out, you'll need to edit it a bit tho:
```python
import time
file = open("CONFIGFILE", "r")
lines = file.readlines()
file.close()
seqtime = lines[0]
lines.pop(0)
def processLine(line: str):
    if line.startswith("#//"): return None
    else:
        return line.strip()
index = 0
while True:
    line = lines[index]
    index = (index + 1) % len(lines)
    line = processLine(line)
    if line == None:
        continue
    else:
        with open("PATHTOFIFO") as fifo:
            fifo.write(line)
    time.sleep(int(seqtime))```

you'll need to replace CONFIGFILE and PATHTOFIFO, for the config file just create a file and then replace the placeholder with the actual path, a example config:
```
5
SOMETHIN
GRADIO
```

the first line is the time in seconds of change and the rest are the PS (you can have as many as you want ps), so each 5 secs the PSes will change, and for the fifo just create one with `mkfifo` and pass it to pifmadv

ValentinVV1, 2023-11-17:
Hi, thank you for the reply!

It unfortunately doesn't work for me. It looks like it's running but does not send anything to the pipe.

Someone sent a bash script on a simillar issue that looks like this:
`#!/bin/bash

while true
do
echo 'PS TEST1' | cat>rds_ctl
sleep 5
echo 'PS TEST2' | cat>rds_ctl
sleep 5
done
exit 0`

That does set the PS to TEST1 but that's it. Removing the sleep command sents the two PS values continously...

closed 2023-11-17 by ValentinVV1

KubaPro010, 2023-11-18:
The code was a bit faulty, as i forgot the "PS" prefix, i've edited the message and it should work for everything

ryangontv, 2023-11-20:
Tried this, doesn't seem to work for me.

I'm running it as a systemd service, could that be the issue?

Ryan

KubaPro010, 2023-11-22:
> Tried this, doesn't seem to work for me.
> 
> I'm running it as a systemd service, could that be the issue?
> 
> Ryan

I've updated the script a bit, type in `sudo journalctl -u {SERVICE_NAME}` and press page down until it doesnt scroll and send the log

ryangontv, 2023-11-22:
Thanks Kuba, I'll look at it when I am at home again.

Ryan