miegl/PiFmAdv #113: too few arguments to function

ShadowinFlight, 2024-09-08:
Device: PI 4B
When I try to build it comes out this:
rds_wav.c:81:13: error: too few arguments to function ‘fm_mpx_get_samples’
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
it seems that the function "fm_mpx_get_samples" is lack of some arguments.
As I check your source code, I found that the function "fm_mpx_open" in file "rds_wav.c" has the same issue.
please fix it.
:)

henrydatei, 2024-12-03:
It seems to be solved by https://github.com/miegl/PiFmAdv/pull/102

zoldaten, 2025-02-13:
**for aarch64 raspberry pi:**
- Makefile should be fixed as https://github.com/miegl/PiFmAdv/pull/102/files
- sudo apt install libsoxr-dev

dhrobinson, 2025-07-30:
I have both libraries installed and the latest code from `master` but get a similar error, in "fm_mpx_open" in file "rds_wav.c"

```
gcc -Wall -Wno-multichar -std=gnu99 -c -O2 rds_wav.c
rds_wav.c: In function ‘main’:
rds_wav.c:53:8: error: too few arguments to function ‘fm_mpx_open’
   53 |     if(fm_mpx_open(in_file, LENGTH) != 0) {
      |        ^~~~~~~~~~~
In file included from rds_wav.c:32:
fm_mpx.h:8:12: note: declared here
    8 | extern int fm_mpx_open(char *filename, size_t buf_len, int cutoff_freq, int preemphasis_corner_freq, int srate, int nochan);
      |            ^~~~~~~~~~~
rds_wav.c:81:32: warning: passing argument 1 of ‘fm_mpx_get_samples’ from incompatible pointer type [-Wincompatible-pointer-types]
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |                                ^~~~~~~~~~
      |                                |
      |                                float *
fm_mpx.h:9:39: note: expected ‘double *’ but argument is of type ‘float *’
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, int mpx_buffer_len, float mpx, int rds, int wait);
      |                               ~~~~~~~~^~~~~~~~~~
rds_wav.c:81:13: error: too few arguments to function ‘fm_mpx_get_samples’
   81 |         if( fm_mpx_get_samples(mpx_buffer) < 0 ) break;
      |             ^~~~~~~~~~~~~~~~~~
fm_mpx.h:9:12: note: declared here
    9 | extern int fm_mpx_get_samples(double *mpx_buffer, int mpx_buffer_len, float mpx, int rds, int wait);
      |            ^~~~~~~~~~~~~~~~~~
make: *** [Makefile:51: rds_wav.o] Error 1
```